prusa3d/PrusaSlicer #9536: On Delta printers, move the purge stripe away from Towers

FlailAway, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
My fan duct is rubbing the belt in the same spot for every print with that purge line. Change 30+ lines of G01 to G02/G03. Regenerate the default-Start GCode based on the **current** build plate diameter.

**Describe the solution you'd like**
Move that arc to a position between the Delta Towers.

**Describe how it would work**
Nothing there to bump into.

**Describe alternatives you've considered**
I tried rewriting the GCode but because it is all G01 moves it is a messy job. IF you used G02/G03 moves instead of a lot of G01, then the User moving the position would be a simple change of just the Radius.

**Additional context**
I tried changing the build plate diameter in Settings but the code was no regenerated. After a Slice, the purge was beyond the edge of the new build plate diameter. Needing to restrict the diameter is a real issue with Deltas for a variety of reasons. Hard coding the Purge curve is beyond a lot of Users, me included.

neophyl, 2023-02-02:
If your printers firmware can handle G02/03 gcodes then you can add them to your start gcode.  That's the whole point of the start gcode block, its completely customisable by the user, and its fixed to what you (or the profile you started with) puts in that field. There is no 'regeneration' its just a block of commands.    Doesn't matter if the rest of the slicing uses G1 or not, only if your printer can.

rtyr, 2023-02-02:
I believe it's already answered by Neophyl. Purge line is defined by the author of the profiles.

closed 2023-02-02 by rtyr