prusa3d/PrusaSlicer #9538: crash when changing text size of emboss

dupoxy, 2023-02-02:
### Description of the bug

crash when changing text size of emboss

### Project file & How to reproduce

1. add shape > text
2. change font height
crash

https://user-images.githubusercontent.com/1143957/216410719-2385659d-fb10-4558-8b80-b5d3ee619003.mp4

[crash.txt](https://github.com/prusa3d/PrusaSlicer/files/10572025/crash.txt)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+MacOS-x64-202301311728

### Operating system

Mac OS X 10.15.7

### Printer model

none

Jony01, 2023-02-03:
Is crash happens only with height? Could you please try to change also depth.

dupoxy, 2023-02-03:
@Jony01  yes crash only with height
 1. depth works
 2. I can change the font it s no effect on this issue.

dupoxy, 2023-02-04:
update
tested with PrusaSlicer-2.6.0-alpha3
added Terminal output:
```
[2023-02-04 10:16:29.295431] [0x000000011b7e5dc0] [trace]   Initializing StaticPrintConfigs
[2023-02-04 10:16:30.505719] [0x000000011b7e5dc0] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 5): image not found
[2023-02-04 10:16:47.924907] [0x000000011b7e5dc0] [error]   Can not process font('Helvetica'), file in path('/System/Library/Fonts/Helvetica.ttc#postscript-name=Helvetica') is not valid TTF.
2023-02-04 10:16:47.987 PrusaSlicer[92071:8765176] -[NSCTFontDescriptor length]: unrecognized selector sent to instance 0x7fb6dc059900
```

Jony01, 2023-02-08:
I can't reproduce it on our MacOS. I tried to fix it blindly by hint from crash-log. Could you please try this: [MacOS build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha3+35-fs-emboss-temp-MacOS-universal-g040f72187-202302080803.dmg)

dupoxy, 2023-02-08:
@Jony01 Sorry to say the provided build did not help.
new [log-PrusaSlicer-2.6.0-alpha3+35-fs-emboss-temp-MacOS-universal-g040f72187-202302080803.txt](https://github.com/prusa3d/PrusaSlicer/files/10687008/log-PrusaSlicer-2.6.0-alpha3%2B35-fs-emboss-temp-MacOS-universal-g040f72187-202302080803.txt)

Jony01, 2023-02-09:
Thanks for test it. With respect to the crash log. I will try to create minimal viable example for wxWidget guys.

Jony01, 2023-02-09:
Could you capture video of [wx_test build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha3+38-fs-emboss-wx-test-MacOS-universal-gfbe068956-202302091102.dmg)
I add button to test minimal viable example inside emboss window.

Don't be scared I add multiple debug message box to be sure why it is happend.

dupoxy, 2023-02-09:
video:

https://user-images.githubusercontent.com/1143957/217858131-fea9bcbd-00d7-4dcf-b4ca-c3173ba2ce89.mp4

log: 
[log-wx_test build.txt](https://github.com/prusa3d/PrusaSlicer/files/10698770/log-wx_test.build.txt)

Jony01, 2023-02-14:
@dupoxy could you please send me your `PrusaSlicer.ini` file.
Folder with current ini file:
[image]

dupoxy, 2023-02-14:
@Jony01 the new `PrusaSlicer.ini`
[PrusaSlicer.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/10735826/PrusaSlicer.ini.txt)

Jony01, 2023-02-14:
@dupoxy : Could you delete `PrusaSlicer.ini` and start this [build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha3+39-fs-emboss-wx-test-MacOS-universal-g9f0b98ffc-202302141847.dmg) and create video for me.

dupoxy, 2023-02-14:
video:

https://user-images.githubusercontent.com/1143957/218853389-1c5c81e9-93f5-493c-91e2-1900b389da1d.mp4

log:
[log-PrusaSlicer-2.6.0-alpha3+39-fs-emboss-wx-test-MacOS-x64-g9f0b98ffc-202302141847.txt](https://github.com/prusa3d/PrusaSlicer/files/10736600/log-PrusaSlicer-2.6.0-alpha3%2B39-fs-emboss-wx-test-MacOS-x64-g9f0b98ffc-202302141847.txt)

npike, 2023-02-18:
I can also reproduce on a 2012 Mac Mini running 10.15.7 if the logs would be helpful.  Crashes on ALPHA4 also.

Jony01, 2023-02-21:
Hi @dupoxy, thanks for cooperation and videos. I prepare [AddressSanitizer](https://en.wikipedia.org/wiki/AddressSanitizer) build to see if appear detects [memory corruption](https://en.wikipedia.org/wiki/Memory_corruption) [bugs](https://en.wikipedia.org/wiki/Software_bug) to test on your machine please download:

 [PrusaSlicer-2.6.0-alpha4+52-fs-emboss-wx-test-MacOS-universal-asan](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha4+52-fs-emboss-wx-test-MacOS-universal-asan-g7a00171d8-202302201539.dmg)
 
Start application from command line with parameter:  `--loglevel=5`
I would like to see application logs.

You should see:
[image]

please hit the button [wxWidget-test] at the beginning of Emboss window

rainell, 2023-02-21:
Logs generated following your instructions
`[2023-02-21 18:48:47.024138] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' 
[2023-02-21 18:48:47.024193] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' by wxString
[2023-02-21 18:48:47.047575] [0x000000011b711dc0] [info]    Undo / Redo snapshot taken: Add Emboss text object, Undo / Redo stack memory: 0MB Resident memory: 824MB; Peak memory usage: 1,060MB
[2023-02-21 18:48:47.075106] [0x000000011b711dc0] [info]    Undo / Redo snapshot taken: Selección-Añadir de la lista, Undo / Redo stack memory: 0MB Resident memory: 827MB; Peak memory usage: 1,060MB
[2023-02-21 18:48:47.160673] [0x000000011b711dc0] [info]    Undo / Redo snapshot taken: Enter emboss gizmo, Undo / Redo stack memory: 0MB Resident memory: 830MB; Peak memory usage: 1,060MB
[2023-02-21 18:48:47.194484] [0x000000011b711dc0] [info]    Same FontNames hash, cache is used. For clear cache delete file: /Users/rainell/Library/Application Support/PrusaSlicer-alpha/cache/fonts.cereal
[2023-02-21 18:48:47.194669] [0x000000011b711dc0] [info]    OS enumerate 271 fonts (+ 0 can't load = 271 fonts) in 32 ms

[2023-02-21 18:48:47.194707] [0x000000011b711dc0] [debug]   Timer 'enumerate_fonts' spend 33ms
[2023-02-21 18:48:47.194736] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' 
[2023-02-21 18:48:47.194782] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' by wxString
[2023-02-21 18:48:47.195010] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' 
[2023-02-21 18:48:47.195059] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' by wxString
[2023-02-21 18:48:47.200559] [0x000000011b711dc0] [info]    Job 'N6Slic3r3GUI6Emboss15CreateObjectJobE' spend 169ms (process 16ms + finalize 153ms)
[2023-02-21 18:48:52.270715] [0x000000011b711dc0] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSCTFontTraitsAttribute</key><dict><key>NSCTFontProportionTrait</key><real>0.0</real><key>NSCTFontWeightTrait</key><real>0.0</real></dict><key>NSFontFamilyAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' is Only face name defined NativeFontInfoDesc.
[2023-02-21 18:48:52.270923] [0x000000011b711dc0] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSCTFontTraitsAttribute</key><dict><key>NSCTFontProportionTrait</key><real>0.0</real><key>NSCTFontWeightTrait</key><real>0.0</real></dict><key>NSFontFamilyAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' is Face and size defined NativeFontInfoDesc.
[2023-02-21 18:48:52.271347] [0x000000011b711dc0] [trace]   '3;90;0;0;43;<plist version="1.0"><dict><key>NSCTFontTraitsAttribute</key><dict><key>NSCTFontProportionTrait</key><real>0.0</real><key>NSCTFontWeightTrait</key><real>0.0</real></dict><key>NSFontFamilyAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' is New created NativeFontInfoDesc.
`
In addition, logs generated when crash
`[2023-02-21 18:49:40.965088] [0x000000011b711dc0] [error]   Can not process font('Helvetica'), file in path('/System/Library/Fonts/Helvetica.ttc#postscript-name=Helvetica') is not valid TTF.
[2023-02-21 18:49:41.236919] [0x000000011b711dc0] [info]    Undo / Redo snapshot taken: Text: Embossed text, Undo / Redo stack memory: 0MB Resident memory: 949MB; Peak memory usage: 1,060MB
[2023-02-21 18:49:41.278009] [0x000000011b711dc0] [info]    Same FontNames hash, cache is used. For clear cache delete file: /Users/rainell/Library/Application Support/PrusaSlicer-alpha/cache/fonts.cereal
[2023-02-21 18:49:41.278197] [0x000000011b711dc0] [info]    OS enumerate 271 fonts (+ 0 can't load = 271 fonts) in 32 ms

[2023-02-21 18:49:41.278221] [0x000000011b711dc0] [debug]   Timer 'enumerate_fonts' spend 32ms
[2023-02-21 18:49:41.278281] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' 
[2023-02-21 18:49:41.278313] [0x000000011b711dc0] [trace]   Load descriptor '3;90;0;0;43;<plist version="1.0"><dict><key>NSFontNameAttribute</key><string>Helvetica</string><key>NSFontSizeAttribute</key><real>13</real></dict></plist>' by wxString
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1439==ERROR: AddressSanitizer: SEGV on unknown address (pc 0x7fff6f282e69 bp 0x7ffeea1c4aa0 sp 0x7ffeea1c4a68 T0)
==1439==The signal is caused by a READ memory access.
[... cut ...]
rax = 0x001dffff51000674  rbx = 0x00006120001354c0  rcx = 0x00007fff8ec68068  rdx = 0x00007fff8ec6c3b0  
rdi = 0x0000608000196220  rsi = 0x00007fff7a7b6d0d  rbp = 0x00007ffeea1c4aa0  rsp = 0x00007ffeea1c4a68  
 r8 = 0x00007fff8ec6bcb0   r9 = 0x0000000000000009  r10 = 0x00007fff51000670  r11 = 0x840fff854d3f8b4c  
r12 = 0x00006120001354c0  r13 = 0x0000000000000002  r14 = 0x000060b0000e9610  r15 = 0x0000000000000001  
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libobjc.A.dylib:x86_64+0x5e69) in objc_msgSend+0x29
==1439==ABORTING
Abort trap: 6
`

dupoxy, 2023-02-22:
video:

https://user-images.githubusercontent.com/1143957/220703441-2d5c81ce-7197-4309-89b7-647bd3462281.mp4

log:
[Uploading PrusaSlicer-2.6.0-alpha4+52-fs-emboss-wx-test-MacOS-universal-asan-log.txt…]()
@Jony01  Thanks for all your work.

Jony01, 2023-02-23:
Hi @dupoxy, Could you re-upload asan logs. It seems that link to logs aim to this issue page.

dupoxy, 2023-02-23:
@Jony01 I edited the previous comment, not sure what happened .

Jony01, 2023-02-24:
I hope, I saw enough in captured video to be able fix it.
Could you confirm that this build solve the crash:
[PrusaSlicer-2.6.0-alpha4+61-fs-emboss-MacOS-universal-g142a21d00-202302231549.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha4+61-fs-emboss-MacOS-universal-g142a21d00-202302231549.dmg)

dupoxy, 2023-02-25:
Sorry to say, the issue is still there.

[PrusaSlicer:2.6.0-alpha4+61-fs-emboss-MacOS-x64-log.txt](https://github.com/prusa3d/PrusaSlicer/files/10830547/PrusaSlicer.2.6.0-alpha4%2B61-fs-emboss-MacOS-x64-log.txt)

rainell, 2023-02-25:
The problem persists.

Jony01, 2023-02-28:
@dupoxy It is sad .. (again at beginning). So, lets try ASAN build with additional application logs: 
[PrusaSlicer-2.6.0-alpha4+90-fs-emboss-MacOS-universal-asan-g6b2c83446-202302281134.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha4+90-fs-emboss-MacOS-universal-asan-g6b2c83446-202302281134.dmg)
Start application from terminal with parameter: `--loglevel=5` and please send me logs.

dupoxy, 2023-02-28:
The new log :
[PrusaSlicer:2.6.0-alpha4+90-fs-emboss-MacOS-x64-asan-log.txt](https://github.com/prusa3d/PrusaSlicer/files/10852980/PrusaSlicer.2.6.0-alpha4%2B90-fs-emboss-MacOS-x64-asan-log.txt)

Jony01, 2023-03-02:
Thanks @dupoxy for logs.
I prepare next build again please start with parameter  `--loglevel=5`
I hope that I fix it,
[PrusaSlicer-2.6.0-alpha4+98-fs-emboss-MacOS-universal-asan-ge428dcc57-202303020702.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha4+98-fs-emboss-MacOS-universal-asan-ge428dcc57-202303020702.dmg)

dupoxy, 2023-03-02:
@Jony01  you did it  🥳 🎉 .
it's working for me.
@npike or @rainell should test to confirm:

> [PrusaSlicer-2.6.0-alpha4+98-fs-emboss-MacOS-universal-asan-ge428dcc57-202303020702.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.6.0-alpha4+98-fs-emboss-MacOS-universal-asan-ge428dcc57-202303020702.dmg)

log:
[PrusaSlicer:2.6.0-alpha4+98-fs-emboss-MacOS-x64-asan-log.txt](https://github.com/prusa3d/PrusaSlicer/files/10872910/PrusaSlicer.2.6.0-alpha4%2B98-fs-emboss-MacOS-x64-asan-log.txt)

rainell, 2023-03-02:
It worked correctly

dupoxy, 2023-03-02:
This can be closed wen fix is publicly available. Thanks

Jony01, 2023-04-04:
Fixed in [version_2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6)

closed 2023-04-04 by Jony01