prusa3d/PrusaSlicer #9539: [REQUEST] Extruder-Specific Maximum Volumetric Rates

Merlin246, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
Volumetric rates, especially for multi-tool printers (like the E3D ToolChanger or the coming Prusa XL), should have extruder-specific maximum volumetric rates. If the printer is configured with something like a V6 on one tool and a SuperVolcano on another, a global maximum volumetric rate will either be extremely incorrect (much higher than the V6's ~12mm^3/s) or leave exceptional amounts of performance on the table (if set to 12mm^3/s, the SuperVolcano can support 110mm^3/s).

**Describe the solution you'd like**
Move maximum volumetric rate setting to each extruder, for a single extruder printer this may be moved to 'General' but for consistency probably keep it in the extruder settings.

**Describe how it would work**
Same way as it works now, however it is applied to each extruder individually, taking the LOWEST of the filament and extruder maximum rates.

**Describe alternatives you've considered**
Creating multiple filament profiles for the same filament for each extruder and setting the global maximum very high. This is tedious and inelegant.

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2023-02-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9337 (which itself is similar to other existing issues lol)

closed 2023-02-02 by rtyr

Merlin246, 2023-02-02:
@neophyl has this been implemented yet? If not, is it planned to be a part of the 2.6.0 release?