prusa3d/PrusaSlicer #9542: BUG: Fail in first Layer Slicing with two difference Nozzel-Size and two Extruders

RS-3D, 2023-02-02:
### Description of the bug

I have described the problem here:
[https://forum.prusa3d.com/forum/prusaslicer/problem-first-layer-slicing-with-two-difference-nozzel-size-multicolorprint/#post-636698]

Is this a development bug or a setting issue?

Can you help my?

### Project file & How to reproduce

[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5 win64 / 2.6 win64

### Operating system

Windows 10

### Printer model

Toolchanger

DragonFinlay, 2023-02-03:
Hi, could you please attach the 3MF for this file, or any file that helps us reproduce the issue?

RS-3D, 2023-02-03:
Yes,I can.
You can simple reproduce: Build a Custem Printer with 1.Extruder 0.4mm Nozzel and 2.Extruder 0.8mm and Slice with two Colors. The setting I adjusted is only the first layer, with the two options of millimeters and percent to switch.
To explain the files:
1."Cube_with_Singelnozzel_0.8mm": Slice with Singelnozzel 0.8mm and 200% First Layer->result is underextrusion in first Layer -> solution is change the first layer to 0.8mm, **always** select the Extruder with 0.8mm Nozzel->this is not a practical real solution

2."Cube_with_0.4_and_0.8_Nozzel__200%_First_Layer": Slice with two diff. Nozzel and 200% first Layer-> result is underextrusion from the 0.8mm Nozzel, the 0.4mm is good

3."Cube_with_0.4_and_0.8_Nozzel__First_Layer_0.8mm": Slice with two diff. Nozzel and 0.8mm first Layer->result is naturally overextrusion from the 0.4mm Nozzel.

As already written in the community,is in my opinion, the percentage should be related to the nozzle diameter and not to the layer height. Or the introduction of separate layer widths for each individual extruder.
Or can I solve this problem with a different setting?

Thanks for your Help.

[Cube_PrusaSlicer_Firstlayer_BUG.zip](https://github.com/prusa3d/PrusaSlicer/files/10582851/Cube_PrusaSlicer_Firstlayer_BUG.zip)

RS-3D, 2023-02-07:
@DragonFinlay 
Hi, have you seen the problem yet?

closed 2026-09-29 by prusaslicer-bot[bot]