prusa3d/PrusaSlicer #9543: Slicing provided 3MF gives an "Access violation"

Misterke, 2023-02-02:
### Description of the bug

Slicing attached 3MF gives an "Access violation".  Disabling Brim makes the violation disappear.

### Project file & How to reproduce

[Master Yoda PrusaSlicer Error.zip](https://github.com/prusa3d/PrusaSlicer/files/10573750/Master.Yoda.PrusaSlicer.Error.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 10

### Printer model

Two Trees Sapphire Plus with Klipper

kubispe1, 2023-02-02:
I can confirm that published 2.6.0-alpha2+win64 is producing "Access violation". 
The fix should be part work progress of today.
- WIN 10 
[image]

lukasmatena, 2023-02-03:
Same backtrace as #9489.

kubispe1, 2023-02-03:
As, I mentioned we fixed it and tested it. In the next release, Access Violation Error should not appear and slicing would be fine.
Closing, Thanks

closed 2023-02-03 by kubispe1