prusa3d/PrusaSlicer #9545: Extra perimeters on overhangs prints on air when external perimeters option is enabled

igiannakas, 2023-02-03:
### Description of the bug

Extra perimeters on overhangs prints on air when external perimeters option is enabled.

To enable successful bridging the extra perimeters on overhangs need to print in reverse order when external perimeters option is enabled.

### Project file & How to reproduce

[Extra perimeters on Overhangs with external perimeters first.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10577222/Extra.perimeters.on.Overhangs.with.external.perimeters.first.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Mac OS 13.2

### Printer model

Ender 3 S1 pro / Klipper

igiannakas, 2023-02-03:
[image]
Also screenshot attached.

igiannakas, 2023-02-09:
Bump. Any updates from the dev team?

Cisien, 2023-02-12:
I also have a model that can repro this issue

[Power Supply Enclosure - Enclosure (Left).zip](https://github.com/prusa3d/PrusaSlicer/files/10715545/Power.Supply.Enclosure.-.Enclosure.Left.zip)

SEIDO-JP, 2023-02-18:
Yeap, me too. 
Check this model in the attachment. At the height of 23.8mm (under first "shelf") with enabled option "Extra perimeters on overhangs" it's printing in the air. 
Look a the screenshot. 

[image]

[AppleWatch_Stand_Tall.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10775080/AppleWatch_Stand_Tall.3mf.zip)

Godrak, 2023-03-06:
This will currently happen, when the anchoring area is not large enough for all the generated bridging perimeters. Because this feature was NOT originally developed to allow printing without supports, but to improve overhang surface quality and hopefully reduce amount of supports needed, it was not perceived as a big issue. 

We will look into it and experiment with reordering of the extrusions in some sensible manner.

igiannakas, 2023-03-07:
> This will currently happen, when the anchoring area is not large enough for all the generated bridging perimeters. Because this feature was NOT originally developed to allow printing without [...]

Understood. If it cannot be easily fixed a bit at ING would be great as this is not easily identifiable when previewing the layers - have to find the overhang layer and check the tool path movement to see that this is happening now.

Godrak, 2023-03-10:
It will be fixed in the next release. The algorithm will choose to print anchored extrusions first. Note that this on its own does not ensure printability, as on complex overhangs the extrusion may be too long or turn extensively, and two anchored endpoints will make little difference there. However, the new version will try to make the path ordering much more sensible.

closed 2023-03-10 by Godrak