prusa3d/PrusaSlicer #9547: Organic support with Top contact Z distance produces errors

Agent-Cain, 2023-02-03:
### Description of the bug

On my first ever try with organic support it seemed like the part I tried to print on top of the branches was a bit to heavy and fall over (this might be some error it self, but not my comcern right now)
[image]
So I tried to re-slice the part and noticed a seriouse gap: 
[image]
That's why I tried to slice it with Top contact Z distance set to 0mm end got this bug: 
[image]
So I closed every task besides PS and tried it again, ending in this error: 
[image]

### Project file & How to reproduce

[Error.zip](https://github.com/prusa3d/PrusaSlicer/files/10578604/Error.zip)
Sliced with 1.8mm Nozzle 0.9mm LH (1.2mm first layer) 
Can't export the settings: 
[image]
So here the settings you might be interessted in:
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Win10

### Printer model

Selfbuild 1mx1mx2m voronlike with afterburner

DerGenaue, 2023-02-03:
Regarding the first image:

Try using `Automatic FDM support painter` as is suggested in the Release notes.
This generates a "cradle" around such sharp overhangs and makes sure they are supported properly.

You can in general use paint-on supports to manually adjust the support points and to make sure everything is supported to your liking.

Agent-Cain, 2023-02-06:
I think this won't come out better, but I'll test it.
[image]

Agent-Cain, 2023-02-06:
> Regarding the first image:
>
> Try using `Automatic FDM support painter` as is suggested in the Release notes. This generates a "cradle" around such sharp overhangs and makes sure they are supported [...]

Nope, `Automatic FDM support painter` didn't solve the first problem, but that is easy to tackle with manual painting. This was more like a small notification that calculation of mass or something could be optimized. 

The real problem, not being able to choose 0 z distance of the support, didn't disappear with alpha3. 
Still getting this error:
[image]

Agent-Cain, 2023-02-20:
This error isn't reproducable with the alpha4.

closed 2023-02-20 by Agent-Cain