prusa3d/PrusaSlicer #9549: Slicing "hangs" with support material extrusion width <0.2

jasonleach-dev, 2023-02-03:
### Description of the bug

When using 0.08mm layer height on my 0.4mm nozzle profile organic supports works fine.  This is with default extrusion of 0.4mm and support extrusion 0.36mm.  All other widths default.

When using the exact same* profile but with a 0.2mm nozzle (*associated extrusion width is altered to 0.2mm / 0.18mm and machine nozzle width of 0.2mm) the slicing hits "70% generating support material" and does not complete.  CPU is still processing at 10-30% usage for 30+ minutes.
Adjusting the support extrusion width to 0.2mm supports generate fine.

### Project file & How to reproduce

[reproduction.zip](https://github.com/prusa3d/PrusaSlicer/files/10579429/reproduction.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 11

### Printer model

Ender 3v2

Cisien, 2023-02-06:
@jasonleach-dev Can you go into detail on your system specs? Myself, and another person are running into excessive slicing times with organic supports (an hour plus).

I7-8700k and an I5 8400

Reproduction:
[DnD_Screen_v4_top.zip](https://github.com/prusa3d/PrusaSlicer/files/10669287/DnD_Screen_v4_top.zip)

Reproduce by slicing on older hardware

jasonleach-dev, 2023-02-07:
> @jasonleach-dev Can you go into detail on your system specs? Myself, and another person are running into excessive slicing times with organic supports (an hour plus).
> 
> I7-8700k and an I5 8400
> 
> Reproduction: [DnD_Screen_v4_top.zip](https://github.com/prusa3d/PrusaSlicer/files/10669287/DnD_Screen_v4_top.zip)
> 
> Reproduce by slicing on older hardware
> 
> It's worth mentioning that automatic supports on this model reduces the slicing time from indefinate to about 3-4 minutes on the 8700k

I am running a Ryzen 5 3600 / 16GB RAM

Cisien, 2023-02-07:
My repro project also repros on a ryzen 5950x (64gb ram)

closed 2026-09-29 by prusaslicer-bot[bot]