prusa3d/PrusaSlicer #9550: [2.6.0 Alpha 2] Crashing Slicing at 90%

psyvision, 2023-02-03:
### Description of the bug

When slicing a fairly simple model PrusaSlicer is crashing at 90%.  I have read some other reports here which suggest it is either supports or sequential printing causing the issue.  This project has neither enabled.  

### Project file & How to reproduce

3MF:

[Enclosure - Part 18.zip](https://github.com/prusa3d/PrusaSlicer/files/10579664/Enclosure.-.Part.18.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2-win64

### Operating system

Windows 10

### Printer model

RatRig VCore 3 Tool Changer

psyvision, 2023-02-03:
Possibly same as https://github.com/prusa3d/PrusaSlicer/issues/9470 - was OK when changing to T0

lukasmatena, 2023-02-06:
This is indeed the same as #9470. One thing the two issues have in common is "TIMELAPSE_TAKE_FRAME" line in custom G-Code. This line is confusing the CoolingBuffer, which mistakes it for a toolchange and crashes after failing to read the new tool number. 2.5.0 did not crash, but the CoolingBuffer was probably fooled as well. I am unable to tell if it was observable or not and how.

The issue will be fixed in the next release, thanks for reporting this.

psyvision, 2023-02-06:
Thanks for the feedback @lukasmatena

terryrobison, 2023-02-12:
I have the same issue when slicing with T1 or T2 or T3.  The default print head T0 is ok.

NyxCode, 2023-02-12:
Same issue here!

lukasmatena, 2023-02-13:
Please, wait until 2.6.0-alpha4 is released and retest after that. The original issue reported by @psyvision should be fixed there, if the rest of you indeed report the same problem (and not a different one that looks the same), it will be gone too.

kubispe1, 2023-02-20:
@psyvision  and others please can you retest the issue with the new alpha? 
Thank you very much for your cooperation.
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4

psyvision, 2023-02-21:
@kubispe1 @lukasmatena Alpha 4 has resolved the issue for me, many thanks team!

lukasmatena, 2023-02-21:
@psyvision Thanks for the update. Closing.

closed 2023-02-21 by lukasmatena