prusa3d/PrusaSlicer #9551: [2.6.0-alpha2] Organic support broke off half way through print.

kingargyle, 2023-02-03:
### Description of the bug

[image]
Attached image shows the problem. I used recommended settings to avoid the branches and avoid crossing permitters. For this print it did not cause a complete failure but it seems the support at the half way point did not have enough structure to hold together.

### Project file & How to reproduce

I can email the project file link but can't publicly upload it. Just need email address to send link too. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2 

### Operating system

Ubuntu 22.04

### Printer model

Any cubic Kobra Neo

bubnikv, 2023-02-06:
I am not sure whether we could help.
It is recommended to enable Z-hop and the new feature "Avoid crossing curled overhangs" as mentioned in our change log.

closed 2023-02-06 by bubnikv

kingargyle, 2023-02-06:
Avoid crossing overhangs was enabled. I'll double check. The main problem seems to be some really thin walls on some organic supports. I saw it again this morning so if I can get some pictures of it I'll attach to this issue as, well.

kingargyle, 2023-02-07:
[image]

[image]

The images attached are additional examples of where the support is to thin. Almost like underextrusion but no where else in the print do I see this only in the organic supports and not all but if it happens it happens at roughly the same point. But not necessarily the same height.

bubnikv, 2023-02-07:
The first image looks like an extrusion problem caused possibly by heat creep (insufficient cooling of your hot end).

kingargyle, 2023-02-12:
@bubnikv I would agree but it only happens on the organic supports, no where else in the print.

freakydude, 2023-04-18:
Maybe this is the reason: 
https://github.com/prusa3d/PrusaSlicer/issues/10226#issuecomment-1513745378

Should we open a new issue?