prusa3d/PrusaSlicer #9553: Ability to mix the type of Supports when painting a support

kingargyle, 2023-02-03:
**Is your feature request related to a problem? Please describe.**
There are some cases where on a model organic supports may be good in one section but the normal snug supports may be better in another. Right now if I add supports it is either all one type or no supports. I can't find an option to indicate the type of support modifier I want at a particular location. 

**Describe the solution you'd like**
Allow ability to selectively say the type you want from available options when either painting or adding support parts. 

**Describe how it would work**
Besides being able to change type of a block. Should be able to also edit the type of support it is using. 

**Additional context**
Picture of model in question. 

[image]

The overhang gaps on right in the model are ideal for organic supports but the bridges in the middle where the support failed might have been better with the traditional supports.

morphias2004, 2025-07-30:
More than two years on and still nothing.

Tupson444, 2026-09-13:
still wanted