prusa3d/PrusaSlicer #9555: Organic Supports - Divide By Zero

jcooperCUSD, 2023-02-03:
### Description of the bug

Selected Print Settings -> Style -> Organic and clicked 'Slice now'

PrusaSlicer has encountered a fatal error: "Divide by zero"
Please save your project and restart PrusaSlicer. We would be glad if you reported the issue

### Project file & How to reproduce

[finger_Splinter_left-organic.zip](https://github.com/prusa3d/PrusaSlicer/files/10580548/finger_Splinter_left-organic.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Microsoft Windows 10 Enterprise 10.0.19044 N/A Build 19044

### Printer model

Ender 5 Pro

dougemon, 2023-02-03:
I also noticed this Divide by Zero issue with Organic Supports, specifically when I changed the "Overhang Threshold" value to 89 or 90.

Zannys, 2023-02-03:
i noticed this occur when a try my profiles, if a switch to any system profile this error not appear

kubispe1, 2023-02-06:
- For Win, it shows this Error
- For Linux, it crashes immediately
- For MAC slices without problems

lukasmatena, 2023-02-18:
Should be fixed in [2.6.0-alpha4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4). Let us know if problems persist.

Zannys, 2023-02-18:
> Should be fixed in [2.6.0-alpha4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4). Let us know if problems persist.

i just tested and same error persist, when i use my profile with organic suporte "PrusaSlicer has encountered a fatal error: "Divide by zero""

kubispe1, 2023-02-20:
@Zannys can you provide us with your problematic .3mf?

mikaeltulldahl, 2023-02-28:
I also encountered divide by zero. it happened when I increased the overhang threshold to 90 degrees.
Settings:
- Version 2.6.0-alpha4+win64
- Windows 11
- organic support 
- support on build plate only
- don't support bridges

kubispe1, 2023-04-13:
We have detected some cases/parameters where divide by zero appears, but this specific issue should be fixed. 
Also, a threshold of 90 degrees should be solved. 
I am closing this issue.

commit 7a873201 (2023-02-18): Organic supports: Fixed division by zero for soluble supports
(with zero separation gap between object and supports)
Fixes #9555
closed 2023-04-13 by kubispe1