prusa3d/PrusaSlicer #9556: [2.6.0-alpha3] Organic Supports missing important unsupported overhangs

dougemon, 2023-02-03:
### Description of the bug

I loaded in a 3D Benchy, and tilted it back 20 degrees. With the new Organic Supports enabled, I am expecting the entire underside to need supports under the dark blue area marked "Overhang Perimeter." But a large overhang isn't generating the supports it would need to succeed. I am attaching the 3mf project of a benchy, as well as a 3mf project of a miniature statue where I encounter the same problem.
I have tried increasing the "Overhang Threshold" to the highest it can go. Manually placing a support enforcer seems to help, but the default behavior seems to be missing important spots.
[image]
[image]

### Project file & How to reproduce

[organic supports benchy overhang unsupported.zip](https://github.com/prusa3d/PrusaSlicer/files/10581260/organic.supports.benchy.overhang.unsupported.zip)
[organic supports DRG Gunner Statue overhang unsupported.zip](https://github.com/prusa3d/PrusaSlicer/files/10581341/organic.supports.DRG.Gunner.Statue.overhang.unsupported.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Original Prusa i3 MK3S+

DerGenaue, 2023-02-03:
Possibly a duplicate of #9459 ?

ThomasRahm, 2023-02-04:
Seems to me like it, with my fixes it would look like this (idk what you need a 85° overhang angle for, but this is what i would expect then):
[image]

dougemon, 2023-02-04:
@ThomasRahm Looks great! Ok, I await the next pre release when this fix is included 👍 probably can close this issue when that happens

kubispe1, 2023-02-13:
Hi @ThomasRahm, thanks for cooperation. Overhang threshold now works fine. Will be part of our next release. 
Closing the issue 
[image]

closed 2023-02-13 by kubispe1