prusa3d/PrusaSlicer #9557: Organic Supports crashing when slicing

techis, 2023-02-03:
### Description of the bug

When I load the models for the hinged top for the Prusa Enclosure and set for Organic supports, upon clicking the slice button I get an error "Access Violation", see screenshot. 
[image]

### Project file & How to reproduce

Open the attached project file.
Set 20% infill, Organic supports, supports on build plate only
Click the slice button.
[EnclosureHingeLid.zip](https://github.com/prusa3d/PrusaSlicer/files/10581823/EnclosureHingeLid.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 10

### Printer model

Prusa i3 MK3s+

lukasmatena, 2023-02-03:
Thanks, reproduced.

bubnikv, 2023-02-06:
Fixed with aaffcbc282d5e67b75b35c175d40010b3bc4a639
Thanks for the report.

closed 2023-02-08 by bubnikv