prusa3d/PrusaSlicer #9558: v2.6.0-alpha Export Plate as STL/OBJ Including Supports no longer available when using only Pad

TheWebMachine, 2023-02-03:
### Description of the bug

In v2.5.x you could add only a Pad to an SLA and export the plate as STL when not using other supports.

Now, in v2.6.0-alpha versions, using only a Pad on the object is not revealing the Export Plate as STL/OBJ Including Supports option anymore. You only get the "...Including Supports" menu item if you actually add full supports to the object.

I believe this to be a bug, as a Pad-only object should be exportable in the same manner as Pad+Supports, as has been the case in previous versions. I rely on this ability specifically to prepare object pads/supports before slicing in another slicer's proprietary format because the other slicer (Anycubic Photon Workshop) has *horrible* support for, well, Supports. haha I prep supports in PrusaSlicer and then export the whole plate for final slicing in Anycubic Photon.

### Project file & How to reproduce

In both v2.5.0 and v2.6.0-alpha:
1. add object to SLA plate
2. add only a Pad (no support tree) to the object
3. then try to use File > Export > Export Plate as STL/OBJ Including Supports

In v2.5.0, you will see Export Plate as STL/OBJ Including Supports is available to select. In v2.6.0-alpha versions, this option is grayed out and unavailable.

Now, use Pad + Supports on the object. You will see both v2.5.0 and 2.6.0-alpha expose the Export Plate as STL/OBJ Including Supports menu item properly.

[d30-wPad.zip](https://github.com/prusa3d/PrusaSlicer/files/10582147/d30-wPad.zip)
**v2.5.0:**
[image]
**v2.6.0-alpha3:**
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+linux-x64-GTK3

### Operating system

Ubuntu 20.04.5 LTS

### Printer model

Anycubic Photon D2

kubispe1, 2023-02-06:
Hi, thanks for the report. looks like a bug really this option is missing there...

kubispe1, 2023-02-16:
Thanks, for your issue. We enabled export again. Closing this issue.
commit: 4c3599fac105c0d6f671a047d43198c42cf3468f
SPE-1431

closed 2023-02-16 by kubispe1