prusa3d/PrusaSlicer #9559: PrusaSlicer on MacOS ventura 13.2 / Apple M1 Pro crash on startup.

zeno4ever, 2023-02-03:
### Description of the bug

Can't start PrusaSlicer GUI on my Mac. Slicing via command line works.

Just to be sure I removed second language from preferances in MacOS, this is no solution.

Specifications on my mac :

Macbook Pro 16"
16 Gb Ram
Apple M1 Pro
Ventura 13.1 & 13.2

[debug_loglevel_5.txt](https://github.com/prusa3d/PrusaSlicer/files/10582178/debug.txt) 

### Project file & How to reproduce

On command line start binary file with './PrusaSlicer'

Expected result to start PrusaSlicer, I get a options to send a crash report to apple.

If wanted I can send text of crash report it to Prusa (don't share it here there might be private info in this file).

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version: 2.6.0-alpha3+MacOS-arm64

### Operating system

MACOS 13.2

### Printer model

Prusa MK3

kubispe1, 2023-02-06:
Hi @zeno4ever, we are not able to reproduce it at the moment. Could be helpful to send also crash report for apple. We will try to investigate it with all clues...
petr.kubis@prusa3d.cz

zeno4ever, 2023-02-07:
Crash report send by email.

zeno4ever, 2023-02-08:
I can start the slicer now ( version PrusaSlicer-2.6.0-alpha3+MacOS-arm64-202302031514 )  No idea what changed why it now works. I will close this issue