prusa3d/PrusaSlicer #9560: Scaling issue while in sliced mode

Shah-Priyanshu, 2023-02-03:
### Description of the bug

When tried to rescale the mode (as shown in the image): 
[image]

After hitting enter it rescales the model to this:
[image]

This happens with almost all models

### Project file & How to reproduce

[Rescale_Error.zip](https://github.com/prusa3d/PrusaSlicer/files/10582220/Rescale_Error.zip)

To recreate this error : 
1. Slice a model
2. Try to rescale it while in the sliced mode.
3. It might now happen in the first try, so you will have to do it couple of times.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3+win64

### Operating system

Microsoft Windows 11 Home

### Printer model

Original Prusa i3 MK3S & MK3S+

kubispe1, 2023-02-06:
This issue is reproducible... We are struggling with the new coordinate sys. approach. Thanks for your issue

RAMilewski, 2023-02-18:
I also see this problem on 2.6.0-alpha3 for MacOS.

kubispe1, 2023-12-29:
I believe this issue has been resolved for some time now. I am closing the issue.
@RAMilewski, could you please confirm that it is working for you?

closed 2023-12-29 by kubispe1