prusa3d/PrusaSlicer #9561: occasional crashes

jomba79, 2023-02-03:
### Description of the bug

the system breaks down when slicing when I resize the material and use a support
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/10582844/Test.zip)

### Project file & How to reproduce

[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/10582848/Test.zip)
1. resize by 50%
2. use support (organic or other one)
3. slice it

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha3 wn64

### Operating system

win11

### Printer model

Ender 3 s1 pro

lukasmatena, 2023-02-03:
Please, provide a 3MF. An STL is not enough. Thanks.

jomba79, 2023-02-03:
[test3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10582939/test3mf.zip)
It is also not working with 3mf file.
I tried it in the 2.5.0 released version and it is working but in 2.6.0alpha3 not working.

lukasmatena, 2023-02-03:
This is not a 3MF from PrusaSlicer. I need you to open PrusaSlicer, set everything so it would crash, and then instead of slicing, select File->Save project. That's the file I need. It will contain the complete configuration you used for slicing that project. The crash depends on something in it.

jomba79, 2023-02-04:
[test3.zip](https://github.com/prusa3d/PrusaSlicer/files/10608140/test3.zip)
Oh, I understand now. :)
Here you are

lukasmatena, 2023-02-06:
Thanks. The crash occurs in the same spot as the one reported in #9562.

jomba79, 2023-02-07:
Thx for your update. Can you solve that issue for final version of prusa slicer?

bubnikv, 2023-02-07:
Fixed with 340b685a0d57248ac3a1b9fed13212200db1d30f

closed 2023-02-08 by bubnikv