prusa3d/PrusaSlicer #9562: PrusaSlicer 2.6.0-alpha3

Garuizbido, 2023-02-03:
### Description of the bug

PrusaSlicer crashes when slicing in Classic mode. Arachne engine works fine, however there are some model were arachne wishes to make impossible movements or simply removes a part of it completely. I wanted to see if the Classic Engine would perform better in this case, but whenever I pick Classic the app crashes. Also, the latest Cura also fails to properly slice the model so seems like Arachne is working fine.

### Project file & How to reproduce

1.Print settings
2. Layers and perimeters
3. Advanced
4. Perimeter generator pick Classic
5. Slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Windows 11

### Printer model

Ender 5-S1 custom profile

lukasmatena, 2023-02-03:
Please post a 3MF, thanks.

Garuizbido, 2023-02-03:
Here is the 3MF, sorry. Also, I confirmed with the release version of PrusaSlicer 2.5.0 and the classic mode works fine. The slice isn't perfect around the bandana but Arachne removes it completely.

[ryu.zip](https://github.com/prusa3d/PrusaSlicer/files/10583206/ryu.zip)

Garuizbido, 2023-02-04:
Would also like to add, it seems that it's only for complex geometry as I sliced a simpler model with the same settings, and worked fine with classic.

kubispe1, 2023-02-06:
I can reproduce it. 
- When the Classis generator is on, it crashes. 
- I discoved that **detect thin wall** makes this trouble. Probably in combination with the model's super thin walls.
- 2.5.0. works fine.

@lukasmatena can you help me to investigate it further more?

lukasmatena, 2023-02-06:
I enclose a simplified 3MF which is usable in Debug mode: [ryu_edited.zip](https://github.com/prusa3d/PrusaSlicer/files/10669892/ryu_edited.zip).

There is a tripping assert which gives a clue: `assert(lslice_idx_min != -1);` in `Layer::sort_perimeters_into_islands`. It seems that there is a perimeter slice which is not recognized as belonging to any of the islands (function `point_inside_surface_dist2` returns max. double for all islands that are tested. I don't know why, but @bubnikv might.

bubnikv, 2023-02-07:
Fixed with 340b685a0d57248ac3a1b9fed13212200db1d30f

kubispe1, 2023-02-09:
Thank you for your cooperation issue was tested. The fix will be part of the next release.

closed 2023-02-09 by kubispe1