prusa3d/PrusaSlicer #9563: Overhang Perimeter generated by "Detect bridging perimeters" does not respect pressure equalizer limit

MGunlogson, 2023-02-03:
### Description of the bug

When enabling both Detect Bridging Perimeters and Pressure equalizer, the max volumetric slope is not respected when entering the detected overhang perimeters. 

[image]

### Project file & How to reproduce

1.     Insert model with bridging perimeters
2.     Enable both Detect Bridging Perimeters and Pressure equalizer
3.     Set both max volumetric slopes to a low number (such as 0,5 mm^3/s^2) for best visibility

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Windows 10

### Printer model

frankenEnder

MGunlogson, 2023-02-03:
possibly related https://github.com/prusa3d/PrusaSlicer/issues/9485

kubispe1, 2023-02-07:
Closing as a duplicate, thanks @9485

closed 2023-02-07 by kubispe1