prusa3d/PrusaSlicer #9569: Organic Supports created off bed.

3Delights, 2023-02-03:
### Description of the bug

When I generate organic supports on my model it puts the end most supports's bases off the print bed/printable area:
[image]

I would expect the supports to be generated on bed, or stay within the slots in the model, and branch to the ends...

Tried all different settings and nothing helps.

### Project file & How to reproduce

Here is a zip with the 3mf file and screen shot in it:  
[Column report.zip](https://github.com/prusa3d/PrusaSlicer/files/10583502/Column.report.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+win64

### Operating system

Windows 10 fully updated 22H2 19045.2546

### Printer model

Original Prusa MK3S+ (no modifications).

kubispe1, 2023-02-07:
This is a similar issue as #849. Brim, skirt, and supports are still generating out of bed.

3Delights, 2023-03-22:
I guessed it was but it still needs fixing and is getting very annoying! I still haven't been able to print this model because of this problem. It's been nearly 5 years now!!

3Delights, 2023-04-20:
Doing a quick search of the source code and doesn't src/libsl3r/BuildVolume (hpp/cpp) provide the functionality to check for the beds boundary?  There must be the sort of code required already available as PS can tell if a model is outside the print area, and it can detect other models and avoid collisions within support structures?

1. Create support line.
2. Does it breach the BuildVolume?
3. If yes delete line, mark path as bad and go to 1
   Else carry on...

hstag, 2023-09-28:
I have the same issue. I couldn't print some models with organic supports because of this limitation.

Is there some workaround for it?

hatesyourface, 2023-10-12:
https://github.com/prusa3d/PrusaSlicer/issues/11164, similar. 

Similar issue on the XL; with a model the full width of the bed the right side had the organic support in front of the model, but on the left side (it's a symmetrical object) the support was generated outside the printable area and I was unable to orient the part in a way that it would be usable. The supports weren't symmetrical, even though the object to print was. Either they need to be generated inside the print area or we need the ability to decide where the "trunk" will originate around the object.

jonaskello, 2023-12-10:
[image]

So close to being able to print this on my Prusa MINI but the support is just a little bit outside. Seems like it could had been generated inside.

3Delights, 2024-07-26:
Still a problem... 6 years and counting!

Jan-Soustruznik, 2024-09-17:
Supports that are generated out of bed in the known issue. https://github.com/prusa3d/PrusaSlicer/issues/9569 and original https://github.com/prusa3d/PrusaSlicer/issues/849

closed 2024-09-17 by Jan-Soustruznik