prusa3d/PrusaSlicer #9570: rafts not working

purplewazard, 2023-02-04:
### Description of the bug

when rafts are used they simply lift the model off the build plate and say the x number of layer are empty rafts do not generate. 

### Project file & How to reproduce

add 1 or more layer for the raft option and the problem will be reproduced

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpah 2 and 3

### Operating system

windows 11

### Printer model

every

JaredTamana, 2023-02-05:
Replicated on 2.6.0-alpha2+win64
[image]

kubispe1, 2023-02-07:
Hi @purplewazard and @JaredTamana , please can you share your 3mf. Do you have organic support on ?

JaredTamana, 2023-02-07:
Sure, though the issue is replicable regardless of what settings or models you have loaded. Simply enabling rafts will allow you to replicate this issue.
[Anker_Fan_Duct_Tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/10675593/Anker_Fan_Duct_Tweaked.zip)

closed 2023-02-08 by bubnikv