prusa3d/PrusaSlicer #9571: first layer over lap

purplewazard, 2023-02-04:
### Description of the bug

the first layer for the organic supports for 2 more more objects overlap

### Project file & How to reproduce

add 2 files to print
turn on supports
i pressed the auto arrange button

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 3

### Operating system

windows 11

### Printer model

every

foreachthing, 2023-02-04:
Sounds like this one: https://github.com/prusa3d/PrusaSlicer/issues/9514

kubispe1, 2023-02-07:
I am closing as a known issue and duplicate problem #316 ... Bypass solution is to merge objects together. It could avoid collisions. 
Thanks

closed 2023-02-07 by kubispe1