prusa3d/PrusaSlicer #9573: UI icons are significantly smaller in 2.6.0

gudvinr, 2023-02-04:
### Description of the bug

Icons across UI in 2.6.0 (both alpha 2 and 3) are small compared to 2.5.0

### Project file & How to reproduce

2.5.0
[image]

2.6.0
[image]

It's only an example, but it is the same for all icons like this. On the panel, in dropdowns, lock icons, expert/advanced markers, etc.

I use KDE Plasma (5.26) on Wayland (thus running under xwayland if nothing is changed in this version).  
I have 2 monitors, each uses fractional scale (1.5 and 1.15).

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3+linux-x64-GTK3

### Operating system

Arch Linux

### Printer model

Artillery Genius Pro

gudvinr, 2023-02-05:
Also slice/export buttons quite small too compared to 2.5.0

gudvinr, 2023-06-14:
This is still an issue with RC2 (both appimage and unpackaged version are affected). Makes configuring and slicing really annoying on hidpi display

Jan-Soustruznik, 2023-06-15:
Hello, @gudvinr , 

thank you for your feedback. Problem appear after wxWidgets update - now wxWidgets manage the icon size by itself... So problem isnt in PrusaSlicer app. We have to contact wxWidgets autors and We try to fix in future. I think we won't to make it in final release 2.6.0.

Issue created: SPE-1785

YuSanka, 2023-06-16:
[Here](https://github.com/wxWidgets/wxWidgets/issues/22807) is related wxWidgets issue

gudvinr, 2023-10-09:
This issue was closed without being fixed (not for GTK at least)

closed 2026-09-29 by prusaslicer-bot[bot]