prusa3d/PrusaSlicer #9574: In the Text embossing tool, clicking the minus button [ - ] of the Height field causes PrusaSlicer crash.

rainell, 2023-02-04:
### Description of the bug

In the Text embossing tool, clicking the minus button [ - ] of the Height field causes PrusaSlicer crash. In addition to the fact that the text that is added is huge.
https://user-images.githubusercontent.com/30863825/216737956-b0ab246f-aeaf-40f2-9a73-c5c9d57c09e1.mp4

[image]

### Project file & How to reproduce

Right click on the Shape-Box, click Add part and click Text; when the Text embossing tool opens, click the minus button [ - ] of the Height field.
[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10593694/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha3

### Operating system

macOS 10.15.7

### Printer model

Any printer (MK3S, Artillery, Ender 3, etc.)

kubispe1, 2023-02-06:
Looks it is specific fot MAC. Win is fine.. Thanks

kubispe1, 2023-02-07:
Can you please provide us crash log generated to be send to Apple ? 
There is my mail petr.kubis@prusa3d.cz to avoid any security issue.

rainell, 2023-02-07:
Of course!

rainell, 2023-02-07:
I sent it to your email

kubispe1, 2023-02-07:
Nice, I have it there...

Jony01, 2023-02-07:
https://github.com/prusa3d/PrusaSlicer/issues/9538

closed 2023-02-07 by Jony01