prusa3d/PrusaSlicer #9575: Add an option to create fillet for embossed text

gudvinr, 2023-02-04:
**Is your feature request related to a problem? Please describe.**

I use embossed text to make molds from plates with, well, text on them.  
It's quite hard to demold from mold if it has sharp edges, especially when it comes to small features (like text).

So when I do that in CAD, I always round both inner and outer edges of the text to make life easier.

I do not mean rounding features of the font itself, only edges that are produced by embossing operation: both inner (between text and body) and outer (between vertical wall and topmost surface) edges, see screenshot below. Similar operation could be done for debossed text, difference is what is outer and what's the inner edge.

**Describe the solution you'd like**

I'd like to be able to round edges of embossed (or debossed) text.

**Describe how it would work**

When you use embossing tool, there could be 2 sliders/input fields to set value for inner and outer edge round separately.

**Describe alternatives you've considered**

Continue making embossed text in CAD.

**Additional context**
[image]

Jony01, 2023-02-07:
Am I correct that you want Bevel for embossed text? 
[image]

gudvinr, 2023-02-07:
I guess so, it's just usually called fillet in CAD software

closed 2026-09-29 by prusaslicer-bot[bot]