prusa3d/PrusaSlicer #9577: Limit the dimensions of the cutting plane in the Cut tool?

rainell, 2023-02-04:
In the example of the attached image, the intention is to cut off the Hulk's arm, but that would mean cutting off the leg as well. 
The problem would be solved if the dimensions of the cutting plane could be limited.
[image]

knightsljx, 2023-04-09:
upvote for this. user should be able to resize the cut plane by stating the length/width of the plane. this will help greatly in cutting complicated parts

kubispe1, 2023-05-17:
Hi, this improvement has been implemented in PrusaSlicer 2.6.0-beta1. I am closing this issue.
Thanks!

closed 2023-05-17 by kubispe1