prusa3d/PrusaSlicer #9579: Post-processing scripts aren't executed on macOS.

DavidPhillipOster, 2023-02-04:
### Description of the bug

Post-processing scripts aren't executed on macOS.
I expect them to execute, as documented in the help text 

Post-processing scripts aren't executed on macOS.

### Project file & How to reproduce

To reproduce:
on macOS, install  
1. https://github.com/DavidPhillipOster/gcodeToThumbnail/releases/download/1.0/gcodeToThumbnail (I put the executable in /usr/local/bin/ ) - verify that it works by running it from the command line. It will print a usage string.

You can compile from source if you wish.

Open any project file where you've asked it to produce a 128x128 PNG .gcode thumbnail (I've attached such a file that has that set, when run on my machine.) (See the screenshots on https://github.com/DavidPhillipOster/gcodeToThumbnail)

[cat.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10607668/cat.3mf.zip)

Slice and save as .gcode. Verify that the resulting .gcode file has a base64 thumbnail near the end of the file by examining it in a text editor.

Note that the .gcode file has the generic document icon when viewed in Finder.

[image]

**This is the bug** had /usr/local/bin/gcodeToThumbnail run as documented, then the .gcode file will have the correct thumbnail icon when viewed in Finder.

To verify, in Terminal, execute:

    /usr/local/bin/gcodeToThumbnail cat_0.3mm_PLA_MK3_2h26m.gcode

Note that the file now has the correct Finder icon.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+MacOS-arm64

### Operating system

macOS Ventura 13.1 (22C65)

### Printer model

i3 MK3

DavidPhillipOster, 2023-02-04:
After the discussion in https://github.com/prusa3d/PrusaSlicer/issues/6322 -this https://github.com/prusa3d/PrusaSlicer/issues/9579 is of low priority. 

At least for the case I care about.

It would be better for me to write a Spotlight extension that enables the Finder to read the thumbnail out of the gcode file without changing the file, so no post processing task from PrusaSlicer is needed.

But it is still the case that post processing steps don't appear to be executed on macOS.

ankurv2k6, 2024-02-23:
Yup - I can confirm the issue - post processing scripts dont seem as executing in 2.7.2-alpha2 either on the mac. Looks like this issue has existed for a while with no resolution

closed 2026-09-29 by prusaslicer-bot[bot]