prusa3d/PrusaSlicer #9580: 2.6 alpha3 Now we can't have VLH on different objects while applying Organic supports to another object

Snuff1eupagus, 2023-02-04:
### Description of the bug

When applying VHL to one Object on the platter with no supports, you can not apply Organic Sup to a different object.

Slicer gives, "Error Variable layer height is not supported with Organic supports."
This was not a problem with alpha2

### Project file & How to reproduce

Add two cube objects or any other two objects, apply VLH to one of the objects and apply organic sup to the other. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha3 

### Operating system

Win 7 & 10

### Printer model

n/a

rcmaniac25, 2023-02-06:
It was added in alpha3 because everyone kept using VLH and organic supports with alpha2 even though the release notes said it wasn't supported, and so people kept creating tickets about VLH and organic supports with alpha2. I expect it will be supported eventually, but not at this moment.

Snuff1eupagus, 2023-02-06:
@rcmaniac25 

I suspected this also, but it didn't read that way in the release summery, and since we have never been able to apply a single instance of VLH to all objects on the platter simultaneously, I felt I should report it as a problem.  Also I thought that those who know enough to work around the issue in alpha 2 would be hopefully smart enough not to report this as an issue.

Thanks for your response.

bubnikv, 2023-02-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9462
Closing this one.

closed 2023-02-06 by bubnikv