prusa3d/PrusaSlicer #9581: Build: Configuration failed: duplicit target - libexpat

mkyral, 2023-02-04:
### Description of the bug

I'm trying to build latest PrusaSlicer on Gentoo and configuration fails detection of libexpat library

```
-- EXPAT found in /usr/lib64/cmake/expat-2.5.0
CMake Error at CMakeLists.txt:448 (add_library):
  add_library cannot create target "libexpat" because another target with the
  same name already exists.  The existing target is an interface library
  created in source directory
  "/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3".
  See documentation for policy CMP0002 for more details.
```

Issue can be solved by deleting/renaming file  `cmake/modules/FindEXPAT.cmake`. So issue is probably there:
```
# Find expat. We have our overriden FindEXPAT which exports libexpat target
# no matter what.
find_package(EXPAT REQUIRED)

add_library(libexpat INTERFACE)

if (TARGET EXPAT::EXPAT )
    target_link_libraries(libexpat INTERFACE EXPAT::EXPAT)
elseif(TARGET expat::expat)
    target_link_libraries(libexpat INTERFACE expat::expat)
endif ()
```

### Project file & How to reproduce

1. run cmake

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Gentoo Linux

### Printer model

MK3S

Salamandar, 2023-03-10:
Same on Archlinux's AUR prusa-slicer-git package.

pmjdebruijn, 2023-06-10:
Happens on Ubuntu 22.04 as well, and the patch seems to resolve the issue, please merge...

commit 71aca4f4 (2024-06-14): Fix problems with finding libexpat

fixes #9581
fixes #10390
closed 2024-06-14 by lukasmatena