prusa3d/PrusaSlicer #9582: Build failed with CGAL 5.5: UserVisitor’ {aka ‘struct priv::Visitor’} has no member named ‘start_filtering_intersections’

mkyral, 2023-02-04:
### Description of the bug

I'm trying to compile latest PrusaSlicer on Gentoo, where only CGAL 5.5 is available.
```
 [I] sci-mathematics/cgal
     Available versions:  5.5(0/14) {doc examples}
     Installed versions:  5.5(0/14)(13:11:18 20.1.2023)(-doc -examples)
     Homepage:            https://www.cgal.org/
     Description:         C++ library for geometric algorithms and data structures
```

Configuration is without issues, but compilation fails on:
```
[121/400] /usr/bin/x86_64-pc-linux-gnu-g++ -DCGAL_USE_GMPXX=1 -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3_build/src/platform -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3_build/src/libslic3r -isystem /usr/include/eigen3 -isystem /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libigl  -DNDEBUG -march=native -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -frounding-math -std=gnu++17 -MD -MT src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libslic3r/CutSurface.cpp
FAILED: src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o
/usr/bin/x86_64-pc-linux-gnu-g++ -DCGAL_USE_GMPXX=1 -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3_build/src/platform -I/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3_build/src/libslic3r -isystem /usr/include/eigen3 -isystem /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libigl  -DNDEBUG -march=native -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -frounding-math -std=gnu++17 -MD -MT src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o -MF src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o.d -o src/libslic3r/CMakeFiles/libslic3r_cgal.dir/CutSurface.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libslic3r/CutSurface.cpp
In file included from /usr/include/CGAL/Polygon_mesh_processing/corefinement.h:23,
                 from /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libslic3r/CutSurface.cpp:28:
/usr/include/CGAL/Polygon_mesh_processing/internal/Corefinement/Visitor.h: In instantiation of ‘void CGAL::Polygon_mesh_processing::Corefinement::Surface_intersection_visitor_for_corefinement<TriangleMesh, VertexPointMap1, VertexPointMap2, OutputBuilder_, EdgeMarkMapBind_, UserVisitor_, doing_autorefinement, handle_non_manifold_features>::start_filtering_intersections() const [with TriangleMesh = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >; VertexPointMap1 = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >; VertexPointMap2 = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >; OutputBuilder_ = CGAL::Polygon_mesh_processing::Corefinement::No_extra_output_from_corefinement<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > >; EdgeMarkMapBind_ = CGAL::Polygon_mesh_processing::Corefinement::Ecm_bind<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Edge_index, bool>, CGAL::Polygon_mesh_processing::Corefinement::No_mark<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > > >; UserVisitor_ = priv::Visitor; bool doing_autorefinement = false; bool handle_non_manifold_features = false]’:
/usr/include/CGAL/Polygon_mesh_processing/internal/Corefinement/intersection_impl.h:1723:42:   required from ‘OutputIterator CGAL::Polygon_mesh_processing::Corefinement::Intersection_of_triangle_meshes<TriangleMesh, VertexPointMap1, VertexPointMap2, Node_visitor>::operator()(OutputIterator, bool, bool) [with OutputIterator = CGAL::Emptyset_iterator; TriangleMesh = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >; VertexPointMap1 = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >; VertexPointMap2 = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >; Node_visitor = CGAL::Polygon_mesh_processing::Corefinement::Surface_intersection_visitor_for_corefinement<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >, CGAL::Polygon_mesh_processing::Corefinement::No_extra_output_from_corefinement<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > >, CGAL::Polygon_mesh_processing::Corefinement::Ecm_bind<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Edge_index, bool>, CGAL::Polygon_mesh_processing::Corefinement::No_mark<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > > >, priv::Visitor, false, false>]’
/usr/include/CGAL/Polygon_mesh_processing/corefinement.h:757:10:   required from ‘void CGAL::Polygon_mesh_processing::corefine(TriangleMesh&, TriangleMesh&, const NamedParameters1&, const NamedParameters2&) [with TriangleMesh = CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >; NamedParameters1 = CGAL::Named_function_parameters<bool, CGAL::internal_np::throw_on_self_intersection_t, CGAL::Named_function_parameters<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Edge_index, bool>, CGAL::internal_np::edge_is_constrained_t, CGAL::Named_function_parameters<priv::Visitor, CGAL::internal_np::visitor_t, CGAL::internal_np::No_property> > >; NamedParameters2 = CGAL::Named_function_parameters<bool, CGAL::internal_np::do_not_modify_t, CGAL::internal_np::No_property>]’
/var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha3/work/PrusaSlicer-version_2.6.0-alpha3/src/libslic3r/CutSurface.cpp:1422:44:   required from here
/usr/include/CGAL/Polygon_mesh_processing/internal/Corefinement/Visitor.h:558:18: error: ‘CGAL::Polygon_mesh_processing::Corefinement::Surface_intersection_visitor_for_corefinement<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Vertex_index, CGAL::Point_3<CGAL::Epick> >, CGAL::Polygon_mesh_processing::Corefinement::No_extra_output_from_corefinement<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > >, CGAL::Polygon_mesh_processing::Corefinement::Ecm_bind<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >, CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> >::Property_map<CGAL::SM_Edge_index, bool>, CGAL::Polygon_mesh_processing::Corefinement::No_mark<CGAL::Surface_mesh<CGAL::Point_3<CGAL::Epick> > > >, priv::Visitor, false, false>::UserVisitor’ {aka ‘struct priv::Visitor’} has no member named ‘start_filtering_intersections’
  558 |     user_visitor.start_filtering_intersections();
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

See attached full log.
I'm not sure, what exactly is wrong and how it could be fixed.

### Project file & How to reproduce

[build.log.zip](https://github.com/prusa3d/PrusaSlicer/files/10608127/build.log.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Gentoo Linux

### Printer model

MK3S

hexane360, 2023-02-13:
I'm running into this too. I believe this is related to this change in the [CGAL 5.5 changelog](https://www.cgal.org/2022/07/15/cgal55/):
> Added more functions in the [visitor of the corefinement based methods](https://doc.cgal.org/5.5/Polygon_mesh_processing/classPMPCorefinementVisitor.html) to track progress.

See also: https://doc.cgal.org/5.5/Polygon_mesh_processing/classPMPCorefinementVisitor.html under "Functions used by corefine() for progress tracking".

Making stubs for these functions should fix the issue.

hexane360, 2023-02-13:
Indeed, adding those stubs to [CutSurface.cpp:1174](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/CutSurface.cpp#L1173) fixes the problem.
To prevent stuff like this, `Visitor` should probably inherit [`Default_visitor`](https://doc.cgal.org/5.5/Polygon_mesh_processing/structCGAL_1_1Polygon__mesh__processing_1_1Corefinement_1_1Default__visitor.html) to get these stubs for free.

mkyral, 2023-02-18:
Thanks, I can confirm that this patch works.

bubnikv, 2023-03-03:
Merged with 65ab83d6a90855212f62bac47cd1f7810cc5bfd3
Thanks