prusa3d/PrusaSlicer #9584: Unable enable build-in downloader

michaloM13, 2023-02-04:
### Description of the bug

Whenever I try to enable built-in downloader I get this error message
[image]

### Project file & How to reproduce

[Downloader error.zip](https://github.com/prusa3d/PrusaSlicer/files/10608378/Downloader.error.zip)

How to reproduce(Slicer is in Czech):
Go to Konfigurace ➝ Průvodce nastavením ➝ Downloads ➝ Allow build-in downloader
Click on Dokončit
Error appears

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.6.0-alpha3+linux-x64-GTK3-202302031527.AppImage 

### Operating system

EndevourOS

### Printer model

Prusa I3 MK3S

kubispe1, 2023-02-07:
Potencional duplicate #9565

michaloM13, 2023-02-07:
> Potencional duplicate #9565

closed 2023-02-07 by michaloM13