prusa3d/PrusaSlicer #9585: Filament option for scaling objects

RudolphRiedel, 2023-02-04:
**Is your feature request related to a problem? Please describe.**
I am printing functional parts with PETG and these do shrink.
This is a property of the material so it may also vary between different brands of PETG, for one manufacturer it may be necessary to scale to 100.4% while for a different brand it may be necessary to scale to 100.8% in order to achieve dimensional accuracy.

**Describe the solution you'd like**
Please add an option to the filament settings, probably under "Advanced" and perhaps as expert option
to scale all objects when printing with this filament.

**Describe how it would work**
The scale-factor in the filament settings is applied to every object upon loading it.
For example if the setup is 100.6% an object on the plater is to be scaled to 100.6% on X/Y/Z on loading
and the "Object manipulation" information displays 100.6%.

**Describe alternatives you've considered**
What I need to do now is to change the scaling for every object I want to print and this is easy enough to forget, even more likely to mess up with different brands of PETG.
The other alternative would be to change the steps/mm in the printer but doing so likely decreases the print quality as it would need to be done using micro-steps and especially Z is prone to loose micro-steps during prints.

neophyl, 2023-02-04:
This looks like a duplicate of several existing issues such as https://github.com/prusa3d/PrusaSlicer/issues/3078 for example.

RudolphRiedel, 2023-02-04:
You are right, this is a duplicate but rather of #4475 for example and not #3078.
The difference is that #4475 is pretty much the same as this one here while #3078 suggests to go a step beyond a static setting and consider the bed temperature for a calculation to compensate for shrinkage.

closed 2023-02-04 by RudolphRiedel