prusa3d/PrusaSlicer #9587: Incorrect detection of print stability issues with simple puzzle cube piece

hippotastic, 2023-02-04:
### Description of the bug

In Version 2.6.0-alpha3+win64 (and Alpha 2 as well), when slicing the attached simple puzzle cube pieces with default settings (original Prusa MK3S+ profile for 0.4mm nozzle, Prusament PLA, 0.3mm DRAFT or any other default print profile), I receive the following warning message for one piece:

```
WARNING:
Detected print stability issues:

Object Body02.stl << !floating object part.

Consider enabling supports.
```

When I pick a default print profile with lower layer height, e.g. 0.05mm ULTRADETAIL, more issues are added to the warning text, namely `!collapsing overhang, loose extrusions`.

However, enabling supports is not necessary at all for this object. When enabling them, the warning disappears, but only two layers of support material get generated which do not provide any actual support.

For comparison, I also included a second piece from the same puzzle in the file (Body05.stl), which has a very similar form, but no warnings get generated for this piece.

Note: Both pieces are taken from the puzzle found here on Printables.com: https://www.printables.com/model/280490-4x4x4-puzzle-cube-hard/files

I'm happy to help in case any further information is needed.

### Project file & How to reproduce

[Puzzle Cube Piece Stability Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/10608694/Puzzle.Cube.Piece.Stability.Issue.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Prusa MK3S+

chriskeeble, 2023-03-11:
I'm seeing similar issues in 2.6.0-alpha5+win64 when slicing very simple, low height 3d objects - e.g. attached screenshot

[image]

bobasaurus, 2023-04-07:
I'm getting a similar error:

[image]

boksbox, 2023-06-24:
Similar warning on Prusa 2.6.0.   Warning goes away when enabling Generat support materials although no supports were created.
MacOS 12.6.6

[image]

suprchunk, 2023-09-29:
I am getting the same "error", but I am adding multiple objects and hand arranging them. If I slice after each addition and move it doesn't affect it. Sometimes it will slice without the error. It is a strange one, that I can replicate all day long.

Item loaded and sliced:
[image]

Multiply item and slice again:
[image]

kendrickxy, 2023-12-28:
Hi team, 
encountering similar problem with the version 2.7.1.

[image]

Paint on supports couldn't find anything with the automatic painting.

tommck, 2024-12-03:
I can't believe this is still open and not addressed. I get this problem ALL THE TIME with almost everything I slice and I'm on 2.8.1 at this point.  Simple things with maximum of a 45 degree overhang... pretty solid base.  I wish we could change the sensitivity of this warning, or just remove it.  It has never been useful to me

closed 2026-09-29 by prusaslicer-bot[bot]