prusa3d/PrusaSlicer #9588: Slicing with verbose gcode causes invalid gcode in the output and in the preview

Misterke, 2023-02-04:
### Description of the bug

Slice attached 3MF.  After slicing preview and set preview mode to "speed".  Notice that a lot of perimeters are displayed as 200mm/s although the settings have way lower speeds set.

Looking at the g-code it seems commands to set the feed-rate get an appendix which causes them to become invalid g-code syntax and the preview apparently mis-parses this as a high speed.  Ex. `G1 F900perimeter`

It seems to me that the verbose g-code setting is adding this appendix without the required comment sign.

### Project file & How to reproduce

See above.  Just slice attached 3MF with verbose g-code enabled.

[Master Yoda PrusaSlicer Bad Gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/10608984/Master.Yoda.PrusaSlicer.Bad.Gcode.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Two Trees Sapphire Plus with Klipper

The-Futur1st, 2023-02-04:
This is because overhang detection is broken
[image]

lukasmatena, 2023-02-06:
Thanks, reproduced. We are looking into it.

kubispe1, 2023-02-09:
Thank you for your cooperation issue was tested. The fix will be part of the next release.

closed 2023-02-09 by kubispe1