prusa3d/PrusaSlicer #9590: Organic supports create a lot of stringing

ivankrsul, 2023-02-04:
### Description of the bug

Organic supports on Prusa Slicer 2.6-Alpha2+win64 create a lot of stringing, much more than it should.  Printing on a Original Prusa i3 MK3S+ 
[image]
[models.zip](https://github.com/prusa3d/PrusaSlicer/files/10609159/models.zip)

### Project file & How to reproduce

[models.zip](https://github.com/prusa3d/PrusaSlicer/files/10609160/models.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.6-Alpha2+win64

### Operating system

Windows 11

### Printer model

 Original Prusa i3 MK3S+

ivankrsul, 2023-02-04:
Better photo of the stringing issue
[image]

ivankrsul, 2023-02-05:
Turns out that this is not a slicer issue.  Sorry.

closed 2023-02-05 by ivankrsul

foreachthing, 2023-02-05:
What was it then?
I"m just curious ;-)

ivankrsul, 2023-02-05:
Apparently it is humidity in the filament.  I live in a very dry place, so this was never an issue, but the past few days it's been raining so the filament became humid.