prusa3d/PrusaSlicer #9592: skipped first layer of different color on multimaterial

petrubecheru, 2023-02-04:
### Description of the bug

I make a simple STL file to test MMU filament change:
[image]

First layer of different color is skipped:
[image]

Expected result (sliced without painting):
[image]

### Project file & How to reproduce

Here are the files:
MMuTip.FCStd  
MMuTip.stl  
MMU-Tip.3mf  
expected_slice.png  
painted_object.png  
sliced_after_painting.png
[skipedLayer.zip](https://github.com/prusa3d/PrusaSlicer/files/10609406/skipedLayer.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3

### Operating system

Ubuntu 22.10

### Printer model

Original Prusa i3 MK3S & MK3S+ MMU2S

halofx, 2023-02-09:
I am seeing a similar issue when using No Sparse Layers, where the first layer is shifted up, and overlaps the second layer. 
Print preview does show the problem. This happening with 2.6 alpha 2, and 2.6 alpha 3. I am on Win10.

0.20 QUALITY setting shown. 5mm boxes, right box raised .5mm.
[image]

0.2 layer is empty. 
[image]

First layer is printed at 0.45mm height.
[image]

Second layer is printed at 0.4mm layer height, and subsequent layers print as expected. 
[image]

kubispe1, 2023-02-16:
Thanks for your issue, the fix will be part of the next release. Closing issue
Commit: 026ca7b3c961660e3003125e792cfb131ffe7498
SPE-1458

closed 2023-02-16 by kubispe1