prusa3d/PrusaSlicer #9593: External Perimeter Speed Not Applied everywhere

CCS86, 2023-02-04:
### Description of the bug

As you can see here, one of the regions is not getting the correct external perimeter speed applied:

[image]

### Project file & How to reproduce

[PS Duct 2.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/10609444/PS.Duct.2.3mf.txt)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 10 x64

### Printer model

UMO+

rtyr, 2023-02-04:
It is printed with small perimeter speed (140mm/s in your case). Expected behavior.

closed 2023-02-04 by rtyr

CCS86, 2023-02-05:
> It is printed with small perimeter speed (140mm/s in your case). Expected behavior.

Ah, yes I see that.

Can this be disabled, or the length of a "small perimeter" be defined?

Even when setting small perimeter speed to 0, it seems to have some effect in slowing down perimeters deemed small.