prusa3d/PrusaSlicer #9594: Dynamic Overhang Speed Causing Erratic Speeds

CCS86, 2023-02-04:
### Description of the bug

Here you can see on the left, without dynamic overhang speeds, we get a nice taper from normal external perimeter speed to the minimum speed defined in the cooling slowdown section.

On the right, dynamic overhang speeds have been enabled, and speeds become erratic. It begins to slow down, then snaps back to a normal external perimeter speed. Further up, it starts applying the internal perimeter speed to external perimeters for some reason.

[image]

### Project file & How to reproduce

[PS Duct 2.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/10609458/PS.Duct.2.3mf.txt)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 10 x64

### Printer model

UMO+

jared596, 2023-02-07:
I have noticed that enabling dynamic overhang speed will cause travel speed to be applied to external perimeters, seemingly randomly throughout the print when there are no significant overhangs to enable the feature.

kubispe1, 2023-02-09:
Your issue was fixed. Will be part of new release.

closed 2023-02-09 by kubispe1