prusa3d/PrusaSlicer #9596: Command line help does not work according to wiki page

julie777, 2023-02-04:
### Description of the bug

prusa-slicer --help-fff does nothing even though the wiki reports it as the correct way to ask for help
prusa-slicer --help does nothing
prusa-slicer with any bad parameters does nothing
prusa-slicer does open the application graphically

### Project file & How to reproduce

Try to run prusa-slicer.ext from command line to show help

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

windows 10

### Printer model

Prusa MK3S+

foreachthing, 2023-02-07:
Try the same with `prusa-slicer-console.exe`.
Then I get this:

<details>
  <summary>Expand Help-FFF</summary>

```
prusa-slicer-console --help-fff
[2023-02-07 14:22:40.866352] [0x00001df0] [trace]   Initializing StaticPrintConfigs
PrusaSlicer-2.6.0-alpha3+win64-202302031521 based on Slic3r (with GUI support)
https://github.com/prusa3d/PrusaSlicer

Usage: prusa-slicer [ ACTIONS ] [ TRANSFORM ] [ OPTIONS ] [ file.stl ... ]

Actions:
 --export-3mf        Export the model(s) as 3MF.
 --export-amf        Export the model(s) as AMF.
 --export-gcode, --gcode, -g
                     Slice the model and export toolpaths as G-code.
 --export-obj        Export the model(s) as OBJ.
 --export-sla, --sla Slice the model and export SLA printing layers as PNG.
 --export-stl        Export the model(s) as STL.
 --gcodeviewer       Visualize an already sliced and saved G-code
 --help, -h          Show this help.
 --help-fff          Show the full list of print/G-code configuration options.
 --help-sla          Show the full list of SLA print configuration options.
 --info              Write information about the model to the console.
 --opengl-debug      Activate OpenGL debug output on graphic cards which support it
 --opengl-version ABCD
                     Select the specified OpenGL version
 --save ABCD         Save configuration to the specified file.
 --slice, -s         Slice the model as FFF or SLA based on the printer_technology configuration
                     value.

Transform options:
 --align-xy X,Y      Align the model to the given point.
 --center X,Y        Center the print around the given center.
 --cut N             Cut model at the given Z.
 --delete-after-load ABCD
                     Delete files after loading.
 --dont-arrange      Do not rearrange the given models before merging and keep their original XY
                     coordinates.
 --duplicate N       Multiply copies by this factor.
 --duplicate-grid X,Y
                     Multiply copies by creating a grid.
 --ensure-on-bed     Lift the object above the bed when it is partially below. Enabled by default,
                     use --no-ensure-on-bed to disable.
 --merge, -m         Arrange the supplied models in a plate and merge them in a single model in order
                     to perform actions once.
 --repair            Try to repair any non-manifold meshes (this option is implicitly added whenever
                     we need to slice the model to perform the requested action).
 --rotate N          Rotation angle around the Z axis in degrees.
 --rotate-x N        Rotation angle around the X axis in degrees.
 --rotate-y N        Rotation angle around the Y axis in degrees.
 --scale N           Scaling factor or percentage.
 --scale-to-fit X,Y,Z
                     Scale to fit the given volume.
 --split             Detect unconnected parts in the given model(s) and split them into separate
                     objects.

Other options:
 --config-compatibility
                     This version of PrusaSlicer may not understand configurations produced by the
                     newest PrusaSlicer versions. For example, newer PrusaSlicer may extend the list
                     of supported firmware flavors. One may decide to bail out or to substitute an
                     unknown value with a default silently or verbosely. (disable, enable,
                     enable_silent; default: enable)
 --datadir ABCD      Load and store settings at the given directory. This is useful for maintaining
                     different profiles or including configurations from a network storage.
 --ignore-nonexistent-config
[... 978 lines cut ...]
                     The diameter of the thinnest branches of tree support. Thicker branches are more
                     sturdy. Branches towards the base will be thicker than this. (mm, default: 2)
 --support-tree-branch-diameter-angle N
                     The angle of the branches' diameter as they gradually become thicker towards the
                     bottom. An angle of 0 will cause the branches to have uniform thickness over
                     their length. A bit of an angle can increase stability of the tree support. (°,
                     default: 5)
 --support-tree-tip-diameter N
                     The diameter of the top of the tip of the branches of tree support. (mm,
                     default: 0.8)
 --support-tree-top-rate
                     Adjusts the density of the support structure used to generate the tips of the
                     branches. A higher value results in better overhangs, but the supports are
                     harder to remove. Use Support Roof for very high values or ensure support
                     density is similarly high at the top. (%, default: 30%)
Wipe options:
 --wipe-into-infill  Purging after toolchange will be done inside this object's infills. This lowers
                     the amount of waste but may result in longer print time due to additional travel
                     moves.
 --wipe-into-objects Object will be used to purge the nozzle after a toolchange to save material that
                     would otherwise end up in the wipe tower and decrease print time. Colours of the
                     objects will be mixed as a result.
```

</details>

Jony01, 2023-02-08:
I add a note.