prusa3d/PrusaSlicer #9597: supports and perimeters: layers printed with 0 (zero) temperature (mmu)

petrubecheru, 2023-02-04:
### Description of the bug

accordind to PS temperature view, one or more layers of the supports (and one perimeter) is printed with the temperature set to zero. This can clog the extruder.
[image]

perimeter (Z 56.20)
[image]

##### Grid supports (3 layers: Z 5.40, 5.55 and 5.75 ):
 Z 5.40
[image]

Z 5.55
[image]

Z5.75:
[image]

Z 56.20
[image]

#### Organic supports (one layer):
[image]

### Project file & How to reproduce

[sheep-zero-temp.zip](https://github.com/prusa3d/PrusaSlicer/files/10609874/sheep-zero-temp.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3

### Operating system

Ubuntu 22.10

### Printer model

Original Prusa i3 MK3S+ MMU2S

lukasmatena, 2023-02-06:
Thanks. This is quite interesting - I see the same behavior in 2.4.2, 2.5.0 and 2.6.0-alpha3. Whatever the cause, it is apparently there for a long time...

vistax64, 2023-03-06:
hi there, this seems to be partially fixed in 2.6 (alpha 4).  what its doing for me at least, is whatever value you change in "first layer", is what gets assigned to color2 temp.  to reproduce it, change the default pla profile for color1 and color2 for first layer = 205 and other layers = 215.  results is color1 is correct temp throughout the print at 215.  but color2 is for the entire print set to 205 (the value of 'fist layer').  

this was driving me crazy in 2.5 (stable release) as it rendered 0 for color2.  but in 2.5 it only is incorrect if your pla profile for mmu is set to exactly 215 for 'other layers'.  if you change it to 214, the temp is correct.  set it to 215, and the temp is 0.  

but as far as 2.6 (alpha 4), this seems to be partially fixed.  i believe if the fix is pulling the value of first layer, then to truly fix this setting the fix to pull the value of whats assigned in 'other layers' will actually fix this. 

i kept getting failed prints on the mmu2s when color2 was showing 0, and i assume 2.6 will fail if any other users use anything other then defaults for first layer and other layers.  

[image]

vistax64, 2023-03-06:
also as an enhancement, if you leave both 'first layer' and 'other layers' at the same temp (in 2.6a4) the temps is correct for both, but only because 'first layer' is set to 215, color2 is ignoring 'other layers' and only pulling from 'first layer'.  but the enhancement would be for blue to only be used for truly 0c, and let the higher temps be more reddish/orange as normal (non mmu) temp behavior, and produce less anxiety thinking the temps may be 0c.  
[image]