prusa3d/PrusaSlicer #9598: Piecewise sequential prints of tall objects for less stringing

ZevEisenberg, 2023-02-05:
Related to #8789

**Is your feature request related to a problem? Please describe.**

I'm printing a couple of tall objects next to each other. This usually results in some light stringing between the two objects. I'd like to use sequential printing to reduce the stringing between objects, but I can't if more than one of the objects is too tall.

**Describe the solution you'd like**

I'd like the option for a sort of hybrid sequential printing mode. It would print, say, 1 cm of model A, then 1 cm of model B, then back to A for another cm, and so on. This would minimize stringing because the nozzle would be moving back and forth between models only once every cm (or whatever the safe increment is). This would be a nice compromise that gets most of the benefit of sequential printing without being limited by model height.

**Describe how it would work**

I don't know if this would just be enabled all the time, effectively eliminating the height restriction on sequential printing, or would be a sub-checkbox or separate "hybrid sequential printing" option. I want the feature, but I don't know the ins and outs of the UI well enough to know where it should live.

**Describe alternatives you've considered**

Currently, my only alternative to limit stringing is to print the parts separately, or to manually group parts so that I pair tall with short ones to slightly reduce the number of total print jobs.

**Additional context**

I appreciate that one possible answer is "you shouldn't have stringing, just tweak your printer settings." But sometimes I just want a quick hack to reduce stringing, and some materials are extra prone to stringing, so it would be useful to have tricks available to minimize it.

zwimer, 2023-06-18:
This would also solve: https://github.com/prusa3d/PrusaSlicer/issues/1095

zwimer, 2023-06-18:
This is a feature that simplify 3D has that I hope carries over: https://www.simplify3d.com/resources/articles/multi-part-printing/

> This is a very useful printing mode that can help improve reliability and print quality, but you may need to re-arrange your models to use this technique. During sequential printing, the software will print multiple layers of a single model before transitioning to the next model. So it might print 30 layers of our pawn model before it moves over to the knight and prints 30 layers of it. This significantly reduces the amount of movement between models which results in a much cleaner surface finish.

Tupson444, 2023-06-18:
Yes, this would be very useful, even if the printer and model geometry allow only e.g. 10 layers at a time - it would mean 10 times less strings which would be easier to clean.

It certainly can't be too complicated to implement, and it could vastly improve the currently limited sequential printing mode (by enabling it to be used in situations where it's currently not possible because of clearance).

ZevEisenberg, 2026-09-02:
I still think this is a good idea. I forget if we now have sequential printing of taller objects due to the enhanced 3D model of the print head(s) in PrusaSlicer, but I think that an option to minimize inter-model moves would be appreciated.

zwimer, 2026-09-02:
It'd also allow printing some arrangements of objects that default sequential printing would not. ones where the head would otherwise knock over already-printed objects

Jan-Soustruznik, 2026-10-20:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16011.

closed 2026-10-20 by Jan-Soustruznik