prusa3d/PrusaSlicer #9600: prusal slicer 2.6 organic supports not reaching object that needs support

jackfrost311, 2023-02-05:
### Description of the bug

using the new organic support tool, when the support generates, it stops 30 layer lines below the object needing the support, which will cause the print to fail.
[test file for gitbub.zip](https://github.com/prusa3d/PrusaSlicer/files/10609997/test.file.for.gitbub.zip)

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

windows 10

### Printer model

prusa m3s+

neophyl, 2023-02-05:
As mentioned on your forum thread variable layer height is not supported for models with organic support (yet).  This is specifically mentioned in the release notes.

bubnikv, 2023-02-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9462
Closing this one.

closed 2023-02-06 by bubnikv