prusa3d/PrusaSlicer #9601: Organic support generation floating objects and to weak by default

Rapterron, 2023-02-05:
### Description of the bug

Hello Prusa Dev Team.

I noticed a misbehavior during organic support generation:

First of all the default support generation is **to weak** and does **not catch** all support required parts of the model(s) or will simply **break during the print** (as the parts become more and more heavier) but more important **parts are generated floating mid air** and would cause a failed print.

Please find below a few screenshots:

[image]
[image]
[image]

This seems to happen more often when I use paint on supports / support enforcers to make the organic support more dense.

[image]

I hope this Feedback is beneficial for you and thank you for your effort creating and maintaining the best Slicer on the market.
- Cheers Christian

`PrusaSlicer
Version:   2.6.0-alpha3+win64
Build:     PrusaSlicer-2.6.0-alpha3+win64-202302031521

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22621), 64-bit edition
Total RAM size [MB]: 34,104MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070 Laptop GPU/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
[... 370 lines cut ...]
</details>
`

### Project file & How to reproduce

Note: This is a premium licensed model I want to print for a customer / friend. I cut it down to avoid any legal issues but the issue is pretty well reproducible 

[Slicer_issue_cut_version_KassadinPart2.zip](https://github.com/prusa3d/PrusaSlicer/files/10610068/Slicer_issue_cut_version_KassadinPart2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha3

### Operating system

Windows 11

### Printer model

MK3s + MMU2s, MINI

DKotekCZ, 2023-02-05:
I am having same diaspointnent, I've tried to print Spartan model, supports really looked good, but after some hours prints supports breaked and hole print was disapointed. Is there some settings how to make organics support morr useable, or will it be fixed till the final release. I am having Alpha2

cyberknet, 2023-02-13:
I am able to reproduce the issue in a model that I designed myself, no issues posting it here. I am also using paint-on supports, but auto-generated by PrusaSlicer.
Paint On Supports:
[image]

Air Print Supports:
[image]

[Tic-Tac-Toe - Ready To Print.zip](https://github.com/prusa3d/PrusaSlicer/files/10724878/Tic-Tac-Toe.-.Ready.To.Print.zip)

darrenmckie, 2023-02-18:
I am also having this issue.

Rapterron, 2023-02-19:
Hello,

I tested again with the latest alpha 4 but the Issue is still present:

I used the former attached project file and sliced it right ahead.
Here are some examples of floating support which will lead to a failed print.

[image]

[image]

[image]

[image]

jrgiacone, 2023-03-27:
I'm still having the same issue in alpha 5, is there any update on this?

Rapterron, 2023-03-27:
Hello,

I can confirm, in alpha 5 the issue is still present.
However I can see a relation to the paint on support feature.

Objects slices without painted support seems to be (mostly) supported correctly, too weak but at least no or way less islands.

Below my latest test:
Sliced with paint on supports:

[image]

[image]

Sliced without paint on support:
The island on the left side (purple marker) was resolved but there is still a half connected island on the left side (red)

[image]

Hope that helps.
Cheers Christian.

seancmonahan, 2023-04-05:
I can replicate the issue with just a slanted slab. It seems it is occurs when many trunks overlap each other and the slicer merges them. (This is still happening with 2.6.0-alpha6.)

I've attached both a demonstration of the issue [simple-case.zip](https://github.com/prusa3d/PrusaSlicer/files/11161187/simple-case.zip), which contains a single, slanted slab, and an example of what I think is happening under the hood [intersecting-cones-demo.zip](https://github.com/prusa3d/PrusaSlicer/files/11161192/intersecting-cones-demo.zip).

[image]

[image]

`intersecting-cones-demo.3mf` simulates what I think the organic support generation does when building trunks. The case which works has 5 cones overlapping in a cross. The case which breaks has 9 cones overlapping in a 3-by-3 grid. I haven't had a chance to dive into the code yet, but I'd be interested in looking.

[image]

[image]

zgita, 2023-04-08:
This is a ring with square cross-section lifted by a hub with spokes.  The ring floats above the supports in the top of the image.   2.6.0-alpha5
[image]

Madcamper, 2023-04-23:
I too have noticed a similar issue. Mine is along with organic support coming from the build plate up onto the model. As it prints through the hole, it then expands but is printing in the air, rather than attaching to the model.

Maybe need an option to choose for support on the model rather than default to build plate or ability to manually set the starting point for supports? Hope this makes sense. Please send me a message if it doesn't.

Overview of how supports are intended:
[image]

Overview of floating Support:
[image]

Close up of floating support side view:
[image]

Close up of floating support top down:
[image]

View from under the build plate where support is starting:
[image]

Eddyg87, 2023-04-29:
Just to add I am also seeing

 this in Alpha 6. If you slice the attached 3mf file you will observe it on the support being generated for the ears.

[image]

[Mammoth.zip](https://github.com/prusa3d/PrusaSlicer/files/11358921/Mammoth.zip)

bobbyh41, 2023-05-13:
I can confirm that this is still a problem in 2.6 beta 1

JesusFreke, 2023-05-28:
I ran into this bug on 2.6 beta 3. I can provide the sliced model with settings, if requested.

cnlson, 2023-06-02:
@bubnikv
@lukasmatena
@enricoturri1966

Are you aware of this issue? Is someone working on it?

lukasmatena, 2023-06-06:
@cnlson We are aware of the issue.

cnlson, 2023-06-06:
Thank you for confirming

kubispe1, 2023-06-07:
as Lukas mentioned, we have it already on our dev list:
**SPE-1556**

sturze, 2023-06-12:
noticed the same issue on my print. printed fine on alpha 6, on recent versions it starts doing this.
[image]
[Project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11725071/Project.3mf.zip)

jrgiacone, 2023-06-20:
I saw the stable 2.6 got released @kubispe1 has this been resolved?

Madcamper, 2023-06-20:
Doesn't appear to have been fixed, at least in my instance. This is a bird feeder base with a few drain holes in the bottom. The supports are for the bridging of the opening where bird seed falls through and out into the tray. See the last screenshot for reference. I would imagine the first support layer after the top layer of the hole would be printed onto the top layer itself and then built up from there. Or print the supports only directly on top of the top layer of the model.

[image]

[image]

[image]

[image]

[image]

AzimuthHeavyIndustry, 2023-07-01:
@jrgiacone Just made the jump from SuperSlicer to PS to get in on those sweet organic supports, and sadly I can confirm that even in the stable release of 2.6 this is still a problem.

[image]

Trying to support a large, mostly flat, slab at an extreme angle seems to evoke the behavior.  It seems like its having trouble with mashing together the bases of multiple small branches.  Ends up leaving large void spaces and just starting to print branch parts in mid air over them.

kubispe1, 2023-07-10:
At the end of version 2.6.0, we made improvements to Organic Supports in order to reduce the occurrence of this issue, but it still appears as isolated islands in the air. It is proving to be more difficult to fix than we originally anticipated. We apologize for the inconvenience.

bonjorno7, 2023-07-16:
I ran into this issue as well, but it ended up printing fine somehow.
[image]
Otherwise great supports though, keep up the good work 👍

westranm, 2023-08-23:
Issue exists today, for sure. Can't be a much simpler test than the attached to see all the "floaters." I have a large area of a print that has a lip like this that succeeds by pure luck that branch density is so high that it manages to fill in for all the debris that is thrown around the inside of my XL.
[Organic Support Test 230822.zip](https://github.com/prusa3d/PrusaSlicer/files/12414721/Organic.Support.Test.230822.zip)

jps4you, 2023-12-15:
bug continues to exist on 2.7.1. sigh....

shwaydogg, 2023-12-30:
Also stumbled upon this issue while using organic supports.

bootron, 2024-02-07:
Also experiencing this issue in 2.7.1.

Lyude, 2024-02-12:
I hope this issue can be fixed soon! I keep running into model after model that would print perfectly fine with organic support, but fails because it generates mid-air islands :(

westranm, 2024-02-12:
Yes, PLEASE get this one fixed. Organic supports are amazing except for this.

AzimuthHeavyIndustry, 2024-02-12:
Issue has been open this long already and they haven't even bothered marking it as a bug yet.  Somehow I don't think they intend to fix it.  Just move over to Orcaslicer if you want functional organic supports.  I gave up on Prusa getting this fixed months ago.

neclimdul, 2024-02-13:
Don't need to rude, Prusa developers work pretty hard on this and their work flows back and forth with OrcaSlicer. Its just a hard problem and they've got a lot of hard problems to solve so they have to make choices.

Unsurprisingly though, I checked and the sample project creates the similar floating supports in OrcaSlicer. So while you might have had better luck, the bug still exists in shared algorithm that needs to be fixed.

[image]

Orca:
[image]
Prusa:
[image]

tinyfluffs, 2024-04-08:
This bug feels like it's a mathematical conflict between the branch diameter, and the branch angle. One parameter wants to widen the base, whilst the other wants to make it thinner as it gets towards the tips. There isn't a lot of wiggle room either side of the window where the sliced model has good support, yet a thick base. I've been able to workaround that to the best of my ability with these settings. Hope it helps someone stumbling into this thread

[image]

[image]

[image]

Savvy0351, 2024-05-24:
I have found the issue to be both auto generated and painted on organic supports. I can however correct the issue with painted on supports. I use the block supports to space out where the support touch the face and it unclutters the trunks.

The issue arises with large areas needing support close to the build plate. The surface to support is too far away for branches to be created so individual trunks are created instead. The organic generation wants to support the whole thing but it is causing trunks to intersect in the center of the perimeter created by the outside of the glob of trunks. Like @seancmonahan pointed out the floating islands predominantly appear at the intersection of 4 cones in a rough square. when 3 cones intersect they can build off each other instead of creating an island.

using a x=50mm,y=50mm,z=10mm box that is an added part to a default cube of 22.5mm(so I can adjust height from the build plate)
- when box is parallel to build plate the distances from build plate that caused the issue are between 3mm and 9mm
- when box is at a 15 degree angle ~3mm to ~10mm
- when box is at a 45 degree angle ~2mm to ~12mm

I did not test any other sizes of faces or non flat faces. although I have seen it replicated on many parts with non flat and varying sized faces.

The box at 15 degrees starting at 3mm from build plate and ending at ~16mm from build plate really show the transition
[image]
[image]

instead of fixing the issue it may be better to change the programing for the range of 2-12mm

PrusaSlicer
Version:   2.7.4+win64
Build:     PrusaSlicer-2.7.4+win64-202404050928

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 16,981MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 526.98)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1660 Ti with Max-Q Design/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled

Lyude, 2024-08-17:
So - this issue is the bane of my existence, it causes such a large number (like, probably over 95%. it's bad) of models to fail for me that I come back to it pretty frequently and always try to think of some arm-chair solution. I don't usually leave more comments, because I've worked in FOSS long enough to know it's usually not helpful.

But. Something very simple about this problem dawned on me that honestly seems it's worth pointing out. The thing that dawned on me: this is not actually much different of a problem then automatic support generation for models in general, and we could treat the organic tree supports as their own 3D model and automatically generate supports for them in the same fashion we're already doing for the model itself then we could just keep repeating the process until no new branches are generated - which would be a sign no mid-air portions exist any longer.

I can see a few problems with this, but none of them seem too difficult to workaround. For starters: a secondary and subsequent branch generation passes would need to do things slightly differently:

* Interface layers would need to be disabled if the thing being supported is itself an interface layer.
* You would want to ensure these passes act as if "support from build-plate" is always enabled, but with a caveat…
* In addition to being able to start a branch from the build plate, you would also want to make it so that a branch can also start from the inside of the perimeters of a branch.
* Also, I would assume we might want to ignore the minimum branch diameter in situations like this.

I hope this comment isn't totally useless, I waffled for a while on whether I should leave this here - but decided that perhaps someone (maybe even myself if I ever find the time to sit down and learn how prusaslicer works!) can see this idea and run with it. Frankly, with how long this issue hasn't been resolved I can't help but feel a solution like this might be a lot easier to implement then fixing the organic tree algorithm. It definitely feels like one of those solutions that seems simple in hindsight, but certainly is not at all obvious beforehand.

LordMelkor, 2025-04-19:
I can confirm that this issue continues to be present in 2.9.2.

Would be great to see it at least acknowledged as a bug. This issue ruined a 24 hour print for me.

sketch12345, 2025-06-15:
I found a work around, sort of.

I added little cubes in prusa slicer (right click - add shape - box). 
I placed multiple ones around the model (on the ground) where I was getting floating supports above.

The cubes prevent the Slicer from starting with huge support bases, which seems to create those floating support islands. The Slicer avoids the cubes and creates more smaller trees.

This will probably not always work, but I was running out of ideas with my model and this actually fixed it. Without the boxes I have about 5 floating islands, now they are all gone.
Hope this can help someone and I really hope they get this fixed soon, it is a nightmare.

[image]
[image]

5jvm0u4, 2025-09-06:
I think if we could disable this cone shape behavior (as a toggle switch), we could get round this problem in some cases when the problem being surface too close to plate.

[image]

aerobug, 2025-12-17:
Wow! No progress in almost 3 years for such a simple issue? I thought developers of Prusa Slicer were somewhat competent. Such a shame!

mgforbes, 2026-01-04:
Confirmed this bug is still present in the current version of the slicer 2.9.4. This is the first time I've run across this behavior, and it's on a large, flat area. Every other time I've used organic supports, it's been on a smaller area supporting something curved or protruding. Nervously watching the print now to see if the interface layer will stay together enough to get through that level.

MTW-Development, 2026-02-27:
I can confirm that the same issue is happening to me in Bambu Studio v2.5.0.66 (currently the newest version) using the "Default" tree style (which seems to generate organic trees).
Switching to "Slim" fixes this issue. "Hybrid" still has trees generating mid air.

lukasmatena, 2026-09-01:
Improved (although not completely fixed in all cases) in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11).

cyberknet, 2026-09-02:
Commenting that as per @lukasmatena yesterday this is better, but not resolved in PrusaSlicer 3.0.0 alpha so that this doesn't get closed as a legacy issue. My tic-tac-toe example is working correctly, but the simple case from @seancmonahan still fails. 

Note: I did have to load Sean's example into PS 2.9.* first to switch to a Prusa FDM printer (Core One+) as PrusaSlicer 3.0 wouldn't load it due to the original sample using a creality ender 3.

Test results attached for reference.:
[image]