prusa3d/PrusaSlicer #9605: Разрез в PrusaSlicer 2.6

kost111233, 2023-02-05:
Долго пользуюсь 2.5, и там очень удобное редактирование деталей, разрезы и т.д.
Многие детали например с крышками, моделирую одним файлом и легко в prusaslicer 2.5 резал их на 2 части, но в prusaslice 2.6 эту возможность отключили. Прошу вас добавить эту возможность вновь.
[image]

YuSanka, 2023-02-06:
> Долго пользуюсь 2.5, и там очень удобное редактирование деталей, разрезы и т.д.
Многие детали например с крышками, моделирую одним файлом и легко в prusaslicer 2.5 резал их на 2 части, но в prusaslice 2.6 эту возможность отключили. Прошу вас добавить эту возможность вновь.
=
>I have been using 2.5 for a long time, and there is very convenient editing of details, cuts, etc.
Many parts, for example, with covers, are modeled in one file and easily cut them into 2 parts in prusaslice 2.5, but in prusaslice 2.6 this feature was disabled. Please add this feature again.

Thanks for your report. But I really don't understand why do you use cut tool instead of "Split to Objects" functionality.

YuSanka, 2023-02-09:
> Да, я знаю про этот способ. Просто привык так :-)
> Yes, I know about this method. I just got used to it :-)
> 
We will continue in other duplicate issue(#9629), which is in English and everyone can understand our conversation.

> Так же хочу заметить что очень не удобно разрезать маленькие объекты, новая шкала почти полностью закрывает деталь
> I also want to note that it is not very convenient to cut small objects, the new scale almost completely covers the part
>
Yes, this issue is already fixed and will be a part of next release

closed 2023-02-09 by YuSanka