prusa3d/PrusaSlicer #9608: Organic Supports first layer pads are too thin, therefore very difficult to remove, PrusaSlicer 2.6 alpha 3.

wa0uwh, 2023-02-05:
Wow, I love the new PrusaSlicer_2.6 alpha 3, the neat new features are a real game changer for me.

Normally I avoid all supports by designing parts to avoid over hangs, but Organic Supports will increase my productivity for special difficult parts.

I only have one real problem, the new Organic Supports sit on a "one layer pad" ( actually, the same as the past original supports that I try to avoid).  On my 3D Printer a one layer "anything" is stuck so hard, that only a very sharp knife can be used to "pick" or "scrape" it off, which can easily damage to platen. Often I have to just leave the previous prints "one layer" things on the platen and then just try to print a new part over them.

Long ago, I originally had a similarly problem with skirts, I normally print a "skirt" on all parts, where I now adjust the skirt to: 3 loops, 5 layers, and min 20mm, otherwise the skirts where also difficult to remove. Thankfully, the skirt options solves my (then noob) removal problem.

I would like a new "pad support option" to set the number of "pad support layers", or I would suggest it a default to at least 3 to 5 layers.

theDiverDK, 2023-02-05:
I have created 2 bugs about this as well.

1. that the pads loosens from the heatbedm tried it today in 3 different prints.
2. the pad merges with the actual print.

bubnikv, 2023-02-08:
Duplicate of #9448, closing this one.

closed 2023-02-08 by bubnikv