prusa3d/PrusaSlicer #9611: `Add modifier` -> `Text` works differently when right clicking on object or in parts list

ulab, 2023-02-05:
### Description of the bug

Adding a text modifier does not work by right clicking on a part on the sheet in `3D editor view`.

### Project file & How to reproduce

Right click on a part on the sheet in `3D editor view` and select `Add Modifier` -> `Text`. 
This just adds a new text part on the bed.

Right click on a part in the parts list and select `Add Modifier`-> Text. 
This correctly adds a text modifier to the part which can be attached to a surface of the part.

Interestingly enough it did not happen when I just added a basic `Box` shape as part

3mf showing the text part and the text modifier:
[Gridfinity E3D Revo Nozzle Holder 1x2x3(1).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10610903/Gridfinity.E3D.Revo.Nozzle.Holder.1x2x3.1.3mf.zip)

The STL file I wanted to add text to:
[Gridfinity E3D Revo Nozzle Holder 1x2x3(1).stl.zip](https://github.com/prusa3d/PrusaSlicer/files/10610905/Gridfinity.E3D.Revo.Nozzle.Holder.1x2x3.1.stl.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

macOS Ventura Version 13.1 (22C65)

### Printer model

Prusa i3 MK3S+ MMU2S

Jony01, 2023-02-07:
Your model is broken. You have to fix it.
https://help.prusa3d.com/article/corrupted-3d-models-for-printing_2205

closed 2023-02-07 by Jony01

Jony01, 2023-02-07:
[Gridfinity E3D Revo Nozzle Holder 1x2x3(1)_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/10674076/Gridfinity.E3D.Revo.Nozzle.Holder.1x2x3.1._fixed.zip)

ulab, 2023-02-07:
1. The model was not reported as broken by PrusaSlicer. 
[image]
2. This does not explain why it works using the sidebar, but not directly clicking on the model?
3. Aspose's repair tool reports "Some parts have reversed normals". After conversion, PrusaSlicer reports it as broken and auto-repairs 2359 errors. The issue stays the same.
[image]
4. Formware's STL repair reports "447 Non-manifold edges, 721 Inverted faces and 324 Duplicate faces", the repaired file works as expected.

I still think this is an issue, because PrusaSlicer did not report it as broken and it "kind of" worked one way, but not the other. This will confuse users.

Jony01, 2023-02-07:
ad1) Sorry, fast detection of broken model is not detecting 100% of issues.
ad2) [Ray Casting](https://en.wikipedia.org/wiki/Ray_casting) did not work on your model.

Fix after your comment: I changed fail recovery behavior to create part (similar as right panel) instead of creation new object, when model is not hitted by ray caster. 

Still, dragging text over surface will not work til you repair the model.