prusa3d/PrusaSlicer #9612: Issues with local coordinate system and multi part objects.

mix579, 2023-02-05:
### Description of the bug

Alpha has the new World/Object/Part coordinate system. When I load a part and scale it, then add a negative volume and try to change the location of the negative volume using the Object coordinate system, the values I type into X, Y, and Z, get replaced by random numbers. In 2.5, it works as expected.

### Project file & How to reproduce

See video. In this video, the Z numbers don't get changed, but I have seen cases when that happened too. At least this scenario is totally reproducible. It doesn't seem to happen unless I scale the original part first.

https://user-images.githubusercontent.com/78620443/216827635-a14bfc32-29b2-43f3-b381-6013302d6aa5.mp4

Load the 3mf, select the negative volume. Select object coordinates, type any numbers in the x, y, z coordinate boxes. Tab out or click elsewhere. They get replaced by apparently random numbers. Do it again, you get different numbers.

[Shape-Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10611341/Shape-Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusaslicer 2.6-alpha3

### Operating system

MacOS 13.2 Macbook Air.

### Printer model

Mini and Mk3S+

etlam262, 2023-02-14:
I have the same problems and maybe a few more useful observations:
- If I center the negative volume on the object using world coordinates and then manually enter object coordinates it works fine the first time but the coordinates get multiplied by the scaling factor of the object. When I then enter the same coordinates I entered before or some different ones they seem to get "random" and much bigger (for a downscaled object).
- When the negative volume is added before the scaling and then the object get scales everything works fine until new object coordinates are entered for the part.
- When a part is added before the scaling and then only the box part gets scaled (so the object scale stays 100%) everything works fine.
- The object can't be upscaled again after adding the negative volume.

lukasmatena, 2023-09-05:
Can you please check this in [2.6.1-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc2)? Thanks.

mix579, 2023-09-06:
Works correctly for me now with the rc2

lukasmatena, 2023-09-06:
@mix579 Thanks for the update. Closing.

closed 2023-09-06 by lukasmatena