prusa3d/PrusaSlicer #9613: Organic tree island loosens from heatbed

theDiverDK, 2023-02-05:
### Description of the bug

In 3 different prints i have seen the 'island' of the tree's loosen from the heatbed while printing.

The item being printed never loosens, only the tree islands.

I switched to Alpha 3 today, i have printed with alpha 2 without seeing this problem, so it might be new to alpha 3.

It would be nice, if the islands could fuse together to form a large grip area to the heatbed, in exchange for a little more printing.

### Project file & How to reproduce

Printing, both hands or the connector from this LEGO man STL:
https://www.thingiverse.com/thing:2936306

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 Alpha 3

### Operating system

OSX Ventura 13.1

### Printer model

Ender 5 plus

rtyr, 2023-02-06:
> It would be nice, if the islands could fuse together

You can already do that by increasing "First layer expansion value"

[image]

Also you shouldn't use interface layers with current alpha version (it's mentioned in alpha2 release notes), this alone can cause issues you are describing.

theDiverDK, 2023-02-06:
Thanks rtyr

I will try that, I am new to Prusaslicer so I am still learning about all the settings.

About the Interface Layers, you mean i should set there to '0' ?
[image]

rtyr, 2023-02-06:
Yes, set it to 0. Interfaces are currently not working well with organic supports.

closed 2026-09-29 by prusaslicer-bot[bot]