prusa3d/PrusaSlicer #9614: Island for tree support merges with actual print in 2.6 alpha 3

theDiverDK, 2023-02-05:
### Description of the bug

As seen on the screen show, the island (raft) from the tree support is merging with the actual print, making it very hard to remove.

### Project file & How to reproduce

[right_arm.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10611564/right_arm.3mf.zip)
[image]
[image]

Slicing the project, and starting to print, the settings are very close to default, 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 3

### Operating system

OSX Venture 13.1

### Printer model

Ender 5 plus

Chris-Epler, 2023-02-05:
Can confirm, was going to print two parts that were close together and the organic support base happily placed itself overlapping with the adjacent model.

[image]
[image]

foreachthing, 2023-02-06:
See here https://github.com/prusa3d/PrusaSlicer/issues/9514

lukasmatena, 2023-02-06:
Closing as a duplicate of #316.

closed 2023-02-06 by lukasmatena